microsoft/vscode-embedded-tools #2: Zephyr RTOS awareness

beriberikix, 2022-03-10:
Type: Feature Request

Hello! I noticed that the extension currently has RTOS data viewer for Azure RTOS & FreeRTOS. It would be great to see support for the [Zephyr RTOS](https://www.zephyrproject.org/) as well. It is a very active open source RTOS under the Linux Foundation, with over 1K contributors, and increasingly growing in popularity. FWIW, OpenOCD has supported Zephyr since [last year](https://github.com/openocd-org/openocd/commit/77b28ced14321fbedb9493729a377856bab28144).

I'm not sure what's needed in order to add Zephyr support, but I'm happy to connect anyone from the MSFT team with the relevant folks on the Zephyr side.

benmcmorran, 2022-03-22:
Thanks for the feedback! This work is currently on our backlog. I'm working on the support and will be sure to reach out if I have any implementation questions about Zephyr.

robotdad, 2022-04-05:
We have an [RC drop](https://github.com/microsoft/vscode-embedded-tools/releases/tag/v0.1.1-rc.0) that provide a Zephyr RTOS view now, please let us know how it works for you.

ejohnso49, 2022-04-06:
> We have an [RC drop](https://github.com/microsoft/vscode-embedded-tools/releases/tag/v0.1.1-rc.0) that provide a Zephyr RTOS view now, please let us know how it works for you.

I'd love to help test a little bit if I can. Any recommended actions I should take with the extension to test things out?

benmcmorran, 2022-04-29:
The RC should be able to display Zephyr threads in the RTOS view. As with other debuggers, projects must be built with at least `-DCONFIG_DEBUG_THREAD_INFO=y`, and additionally `-DCONFIG_INIT_STACKS=y` and `-DCONFIG_THREAD_STACK_INFO=y` for stack usage information. I was successfully able to test using the `threads` Zephyr sample project.

Other Zephyr object types (e.g. mutexes, queues, etc.) are not currently supported, but are on our backlog.

robotdad, 2022-04-29:
@ejohnso49 were you able to give this a try? Even a "works for me" would be useful feedback. If you weren't able to get it working perhaps the note Ben just posted will help.

robotdad, 2022-04-30:
[vscode.zip](https://github.com/microsoft/vscode-embedded-tools/files/8595312/vscode.zip) contains a preconfigured launch.json and supporting files to get debug working in VS Code. After building from the cli with the proper flags you should be able to adapt this to your needs to launch/debug.

trond-snekvik, 2022-04-30:
Works well for me, at least!

closed 2022-09-29 by benmcmorran