microsoft/vscode-embedded-tools #6: Peripheral View fails to load

sullivanmj, 2022-04-04:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 10 Pro, version 21H2, OS build 19044.1586
- VS Code Version: 1.66.0
- Embedded Tools Extension Version: v0.1.0 Preview
- Target embedded device: Maxim MAX32666GWPBT+T
- RTOS version: FreeRTOS v10.2.0
- Other extensions you installed (and if the issue persists after disabling them): Arm Assembly, Better C++ Syntax, C/C++, C/C++ Extension Pack, C/C++ Themes, C#, Diff Folders, Doxygen Documentation Generator, ESLint, GitHub Repositories, GitLens, Hex Editor, LinkerScript, Pylance, Python, Remote - Containers, Remote - SSH, Remote - SSH: Editing Configuration FIles, Remote - WSL, Visual Studio IntelliCode
- A clear and concise description of what the bug is.

The problem is that the embedded register "peripheral view" is not initializing correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**

The peripheral view should initialize.

**Code sample and logs**
- Code sample
- launch.json:
`{
    "configurations": [
        {
            "name": "GDB",
            "type": "cppdbg",
            "request": "launch",
            "program": "${workspaceFolder}/build/${config:target}.elf",
            "args": [],
            "stopAtEntry": true,
            "cwd": "${workspaceFolder}",
            "environment": [],
            "externalConsole": false,
            "MIMode": "gdb",
            "linux": {
                "miDebuggerPath": "${config:MAXIM_PATH}/Tools/GNUTools/bin/arm-none-eabi-gdb"
            },
            "windows": {
                "miDebuggerPath": "${config:MAXIM_PATH}/Tools/GNUTools/bin/arm-none-eabi-gdb.exe"
            },
            "miDebuggerArgs": "./build/${config:target}.elf",
            "preLaunchTask": "flash",
            "setupCommands": [
                {
                    "description": "Launch OpenOCD from within GDB via pipe connection",
                    "text": "target remote | openocd -c \"gdb_port pipe;log_output openocd.log\" -s ${config:MAXIM_PATH}/Tools/OpenOCD/scripts -f interface/${config:debugger}.cfg -f target/${config:target}.cfg",
                    "ignoreFailures": false
                },
                {
                    "description": "Reset",
                    "text": "monitor reset halt",
                    "ignoreFailures": false
                }
            ],
            "svdPath": "${workspaceFolder}/src/MaximSDK/Libraries/CMSIS/Device/Maxim/MAX32665/Include/max32665.svd"
        },
    ]
}`
- Logs from the Embedded Tools Output Window 
I removed a sensitive portion of path, replaced it with "workingFolder". I also removed the svdString, which is just a printout of the svd file.
`022-04-04T19:00:59.813Z [info] [vscode-embedded] Activating extension
2022-04-04T19:01:29.286Z [debug] [vscode-embedded.webview.registerView] VsCodeHostAdapter [webview -> host]: {"type":"initializeRegisterViewer","messageId":0}
2022-04-04T19:01:29.287Z [debug] [vscode-embedded] Active debug session {"name":"GDB","type":"cppdbg","id":"3b4a5151-385f-4146-9c72-189df9b82350","configuration":{"name":"GDB","type":"cppdbg","request":"launch","program":"workingFolder/build/MAX32665.elf","args":[],"stopAtEntry":true,"cwd":"workingFolder","environment":[],"externalConsole":false,"MIMode":"gdb","miDebuggerArgs":"./build/MAX32665.elf","preLaunchTask":"flash","setupCommands":[{"description":"Launch OpenOCD from within GDB via pipe connection","text":"target remote | openocd -c \"gdb_port pipe;log_output openocd.log\" -s C:/MaximSDK/Tools/OpenOCD/scripts -f interface/cmsis-dap.cfg -f target/MAX32665.cfg","ignoreFailures":false},{"description":"Reset","text":"monitor reset halt","ignoreFailures":false}],"svdPath":"workingFolder/src/MaximSDK/Libraries/CMSIS/Device/Maxim/MAX32665/Include/max32665.svd","__configurationTarget":5,"miDebuggerPath":"C:/MaximSDK/Tools/GNUTools/bin/arm-none-eabi-gdb.exe"}}
2022-04-04T19:01:29.300Z [debug] [vscode-embedded] Workspace configuration {"configuration":{"loggingVerbosity":"debug"}}
2022-04-04T19:01:29.377Z [warn] [vscode-embedded.webview.registerView] String substitution is not supported VsCodeHostAdapter [webview <- host]: {"type":"initializeRegisterViewer","messageId":0,"payload":{"svdString":"...","localization":{"displayLanguage":"en","displayLanguageKind":"VSCode"},"loggingSeverity":"debug"}
2022-04-04T19:01:29.419Z [debug] [vscode-embedded.webview.registerView] Initialized logging
2022-04-04T19:01:29.419Z [debug] [vscode-embedded.webview.registerView] Initializing localization
2022-04-04T19:01:29.420Z [debug] [vscode-embedded.webview.registerView] Loading resources from lang=en-US, ns=resources
2022-04-04T19:01:29.420Z [debug] [vscode-embedded.webview.registerView] Loading resources from lang=en, ns=resources
2022-04-04T19:01:29.420Z [debug] [vscode-embedded.webview.registerView] set register view display language: lang=en, code=VSCode
2022-04-04T19:01:29.550Z [error] [vscode-embedded.webview.registerView] Failed to initialize register view {}`
- Logs from the Debug Console 

**Screenshots**

**Additional context**

Attached file is the contents of max32665.svd: 
[max32665.txt](https://github.com/microsoft/vscode-embedded-tools/files/8412942/max32665.txt)

aleun, 2022-04-07:
Thanks for the bug report, @sullivanmj! I took a look at your SVD file, and it looks like it's got two places where it's not following the CMSIS-SVD spec, so the SVD parser is rejecting them.

I've manually fixed up the SVD, but I don't have your hardware on hand to test it out. Can you try it and let me know if it's working for you? Here's the patch: [max32665.patch.txt](https://github.com/microsoft/vscode-embedded-tools/files/8437382/max32665.patch.txt)

sullivanmj, 2022-04-07:
That fixed it. Thanks a million for the quick response and for fixing that file for me. If there's anything I'd request at this point in terms of a code change, it would be an error message telling me essentially what you've just told me.

Thanks again.

aleun, 2022-04-07:
Glad that worked!

> If there's anything I'd request at this point in terms of a code change, it would be an error message telling me essentially what you've just told me.

I agree that we should be exposing this better. We'll make sure to add better error messages to our backlog.

closed 2022-04-07 by aleun