microsoft/vscode-embedded-tools #7: Memory viewer

joelguittet, 2022-04-05:
Type: Feature Request

I have not tried the extension yet, but one missing feature I do not see in the presentation is the Memory Viewer (and I don't know extension of VSCode proposing this).

This would be a great feature to permit viewing/editing the memory. IDEs from almost all MCU vendors proposes this feature since a long time.

joelguittet, 2022-06-06:
Hello @benmcmorran
Thanks. Is it possible to read lock of memory?
I mean RAM area, this is so much useful when debugging!
Joel

benmcmorran, 2022-06-06:
It's possible to read blocks of memory in some cases. If the block you want to read starts at a register or RTOS address, you can simply click the link to start reading the block. For example, this a ThreadX thread control block followed by the thread's stack, which consists mostly of `0xEF` fill bytes.

[image]

However, VS Code currently doesn't provide a way to read at arbitrary addresses. It's possible to set a watch at an arbitrary address, but there's no option to view the binary data for the address (unlike variables displayed in the Variables view).

[image]

Also notice that the memory view can be difficult to read because the display always starts at address zero, regardless of the real base address. We plan to address these issues in the future.

hongshui3000, 2022-06-08:
In addition to viewing memory, please also support editing memory. This is useful for debugging @[benmcmorran](https://github.com/benmcmorran)

joelguittet, 2022-06-08:
+1 for editing memory, this is a functionality available in other IDE since a long time and it is very very useful!

benmcmorran, 2022-06-09:
Noted. We plan to include support for editing in any memory experience we build.

hongshui3000, 2022-08-29:
Memview:
https://github.com/Marus/cortex-debug/issues/729
https://github.com/haneefdm/memview

haneefdm, 2022-10-13:
Our new repo is https://github.com/mcu-debug/memview

robotdad, 2022-10-20:
We are leaving this request open as we expect to do something here eventually. For now we recommend people use the extension from the repo @haneefdm referenced above.