microsoft/vscode-embedded-tools #13: allow to view other Zephyr objects

hongshui3000, 2022-07-28:
Type: Feature Request

Currently the extension only supports viewing zephyr threads, can it support viewing other objects in zephyr?

benmcmorran, 2022-10-10:
Support for other Zephyr objects is now available in the pre-release channel of the Embedded Tools extension as long as you have version `0.3.221010001` or later installed. As mentioned in the release notes:

> All Zephyr objects supported by Zephyr's object tracking system (timers, memory slabs, semaphores, mutexes, stacks, message queues, mailboxes, pipes, and queues) are now supported in the RTOS view. The `CONFIG_TRACING` and `CONFIG_TRACING_OBJECT_TRACKING` options must be enabled at build time for the Zephyr project, and the `CONFIG_MEM_SLAB_TRACE_MAX_UTILIZATION` option can optionally be enabled to provide additional information for memory slabs. Waiting threads are only displayed when `CONFIG_WAITQ_SCALABLE` is disabled.

We plan to ship this support to the official release channel in the coming days.

benmcmorran, 2022-10-13:
This functionality is now available in version 0.4.0 of the extension, which was released earlier today to the VS Code Marketplace.

closed 2022-10-13 by benmcmorran