microsoft/vscode-embedded-tools #14: Issues with memory views/windows, they are snapshots

haneefdm, 2022-07-28:
Type: Bug Report

**Describe the bug**
- OS and Version: MacOS 12.4
- VS Code Version: 1.69.2
- Embedded Tools Extension Version: v0.1.220728001
- Target embedded device: STM32F429ZI
- RTOS version: FreeRTOS. don't know the version
- Other extensions you installed (and if the issue persists after disabling them):
- A clear and concise description of what the bug is.

Several issues with memory views/windows launched from the RTOS view. Most of them are probably inherent to core VSCode, but I wanted to highlight some that can be perhaps be addressed (or forwarded) here.
* They are never updated. They seem to be a static snapshot. This would be very unexpected of a memory view even if the intent was different. The expectation would be that it updates whenever the session pauses at minimum
* The memory views/windows created demand too much memory. On embedded devices, this can be a problem
  * Some devices are not big. They can have errors in reading memory. Asking for 1-2K is enough
  * If/when memory views are updated upon a pause, it will be bad for performance and UX. Once such a large gdb request is made, it cannot be aborted and further instructions will block. Makes single stepping intolerable

**To Reproduce**
Steps to reproduce the behavior:
1.  In your launch.json, add the enable engine logging as shown below for `cppdbg`
```JSON
            "logging": {
                "engineLogging": true,
                ...
            },
```
2. Launch any debug session, run and pause until an RTOS is displayed by this extension
3. Click on any link in the RTOS view that will open a memory view. I use one of the stack pointers. You will notice in the debug console that a read memory request is made for 128KB (131072 bytes). Here are all the memory reads
```
1: (480376) <-1171-data-read-memory-bytes 0x2000657c 131072
```
4. Single step in any way until you get back into a paused state
5. You will notice in the Debug Console that if no new memory requests were made. I can change memory contents manually using gdb/programmatically and this window never refreshes
6. FYI: Same applies if you launch the memory viewer from anywhere -- as in Watch/Variable panels.

**Expected behavior**
- Any memory views created should be updated just like how Watch/Variable windows are updated. Including REPL. Memory windows are supposed to be alive. They should have a lifecycle of their own.
- I did notice that clicking on the link again will refresh the display, no one would know that
- Please don't read 128KB. 1-2KB is enough, updated with scrolling -- like how disassembly works
- Memory views should be persistent. What an embedded developer expects The lifecycle of these memory windows should be re-thought.

**Code sample and logs**
- Code sample
- `launch.json`
- Logs from the Embedded Tools Output Window 
- Logs from the Debug Console 

**Screenshots**

**Additional context**
I can duplicate all this with the Cortex-Debug debugger as well.

Maybe I am doing something wrong.

haneefdm, 2022-08-28:
Now, I realize that this *probably* has nothing to do with this extension. Its is the way VSCode and the Hex Editor are designed to work with each other. Not sure I can transfer this to somewhere else or what benefit that would provide.

Regardless, there are a lot of issues with Hex Edit and VSCode for memory viewing. I don't think the focus should be on the `editor` part. It should be for efficiency, viewing and debugging experience.

I have an extension in preview right now that you might be interested in and supporting it as a choice for users to say which hex view/edit they want to use.

https://github.com/haneefdm/memview

We can close this but we should continue this conversation (elsewhere) as it is important to the embedded folks. I know my extension is very RAW but try it out and compare it to the Hex Editor experience. I know Hex Editor is much better in many ways but it does not play well with debuggers.

robotdad, 2022-10-20:
We're going to mark this one as won't fix as conversations continue around the memory view extension you have created @haneefdm and leave #7 open to track this longer term.