microsoft/vscode-embedded-tools #16: FreeRTOS Tasks not visible in RTOS viewer tab

Spaark84, 2022-08-12:
Type: Bug Report

- OS and Version: Windows 10
- VS Code Version: 1.70.1
- Embedded Tools Extension Version: 0.2.0
- Target embedded device: STM32WB55
- RTOS version: FreeRTOS 10.2.1
- Other extensions you installed (and if the issue persists after disabling them): IAR Build 1.10.4 and IAR C-SPY Debug 1.10.3

The FreeRTOS tasks of my program are not visible in the Embedded Tools RTOS viewer tab. First, after downloading the program, the tasks list is empty, which is fine since the scheduler has not started yet.

[image]

Then, after running the program and pausing it a while after the scheduler has been started, the task list displays the error message "An error occurred while loading details. Make sure the debugger is paused and try again". 

[image]

My program runs on an STM32WB55 and I'm using the IAR C-SPY debugger with the IAR C-SPY Debug extension for VS code and a STLINK-V3MINI debug probe.

This could also  be an issue of the IAR C-SPY Debug extension.

benmcmorran, 2022-08-15:
@Spaark84 do you still see this issue if you switch to the prerelease version of the extension (version ? I recently [made a change to the FreeRTOS view](https://marketplace.visualstudio.com/items/ms-vscode.vscode-embedded-tools/changelog) to support versions of FreeRTOS that do not include `uxTopUsedPriority`, which might be happening in your case.

[image]

If that doesn't work, try setting `"vscode-embedded.loggingVerbosity": "debug"` in `settings.json`, reproduce your issue, and then attach the content from "Embedded Logging" in the output pane after removing any sensitive information.

Spaark84, 2022-08-16:
@benmcmorran, the problem still exists after switching to the pre-release version (v0.2.220816001). Please find attached the Embedded Logging output. Note that line 1 to 6 of the file are huge as they contain the initialization of the peripheral view.

[Embedded_Logging.log](https://github.com/microsoft/vscode-embedded-tools/files/9350498/Embedded_Logging.log)

I also updated my ST-Link V3 to the latest version v3.10.3, but that didn't help either.

benmcmorran, 2022-08-17:
Thanks for the log. I was able to narrow this down to an issue with the way we were reading results from CSpy. I have an internal PR open with a fix, and it should be available on the pre-release channel within a few days. I'll update this thread once it's available.

benmcmorran, 2022-08-18:
@Spaark84 the fix should now be available in the pre-release channel (version v0.2.220818001 or later). Let me know if it works for you. If not, please attach another log. Thanks!

Spaark84, 2022-08-19:
@benmcmorran, it now works, many thanks.

[image]

closed 2022-09-09 by gcampbell-msft