microsoft/vscode-embedded-tools #17: Use proper columns in peripheral view

Spaark84, 2022-08-12:
Type: Feature Request

Please display the contents of the peripherals side view in a proper columns for register name, value, address, access (r/w) and description and give the user the option to sort the columns and adjust their width.

robotdad, 2022-10-20:
We've discussed this and have decided against it. What you propose is reasonable, but to accomplish it we would need to reimplement our current tree view as a table view. From our perspective the features a table view would provide are nice, but a tree view is more appropriate for peripheral registers. The search and pinning features hopefully provide enough capability to find the registers you need.

Spaark84, 2022-10-21:
In IAR Embedded Workbench, which I currently use, the registers view is a combination of a tree view and a table view, which I think is optimal. However, of course I understand your decision if it is not possible to realize that in VS Code.

[image]