microsoft/vscode-embedded-tools #18: Error message

orzcyt, 2022-08-13:
Type: Bug Report

**Describe the bug**
- OS and Version: ubuntu&windows10
- VS Code Version: 1.70
- Embedded Tools Extension Version: 0.2.220805001
- Target embedded device: STM32
- RTOS version: RTX5
- Other extensions you installed (and if the issue persists after disabling them):
- A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Press F5 into debug mode.
2. Press F6 pause when code is running.
3. See error every time.

**Expected behavior**

It doesn't happen again when I disable Embedded Tools.

**Code sample and logs**
```
Undefined command: "_tx_thread_created_ptr".  Try "help".
Undefined command: "pxReadyTasksLists".  Try "help".
Undefined command: "_kernel_thread_info_offsets".  Try "help".
```

**Screenshots**

[image]

**Additional context**

benmcmorran, 2022-08-15:
@orzcyt thanks for the report. It looks like the Embedded Tools extension is attempting to determine which RTOS is active, but the underlying debugger is not correctly interpreting the request to evaluate the variables and displays an error message. Are you using Cortex-Debug as your debugger? If so, can you switching to the pre-release version of Cortex-Debug? [A change in version 1.5 of Cortex-Debug](https://github.com/Marus/cortex-debug/pull/648/files#diff-8b74b7f531412cc4d779ef482587f0fe9f68c718979c9d3032990457368e2741) improved handling of these evaluate requests.

If that doesn't work, try setting `"vscode-embedded.loggingVerbosity": "debug"` in `settings.json`, reproduce your issue, and then attach the content from "Embedded Logging" in the output pane after removing any sensitive information.

orzcyt, 2022-09-10:
Thank you for your answer.
When I updated the latest cortex-debug & embed tools,the error never happened again.

closed 2022-09-10 by orzcyt