microsoft/vscode-embedded-tools #19: SVD view : Write-only register bits cannot perform write operations

hongshui3000, 2022-08-31:
[image]

gcampbell-msft, 2022-08-31:
@hongshui3000 Could you provide some more information about the situation where this occurred so that we can repro? 

For example, was the debugger in a paused state when this occurred?

hongshui3000, 2022-09-01:
> For example, was the debugger in a paused state when this occurred?

yes，Only bit writes don't work (I click with the mouse, and a write box doesn't show up), the entire register at the top level is writable

hongshui3000, 2022-09-20:
The read-write register can open the write interface
[image]
write-only register, I still can't open the bit write interface here
[image]

gcampbell-msft, 2022-09-20:
@hongshui3000 Correct, this is by design and by nature. 

Bit writes are performed by reading the register value, modifying it, and writing. Due to that, bit writes (also known as field writes), can only be performed on read-write registers. 

Notice that there are no permissions listed beside the bit that you attempted to write. This is because that bit (aka. `field`) is not writable.

benmcmorran, 2022-10-13:
The change to the UI is now available in version 0.4.0 of the extension, which was released earlier today to the VS Code Marketplace.

closed 2022-10-13 by benmcmorran