microsoft/vscode-embedded-tools #20: Plugin cannot detect STM32CubeIDE

Michal123456747, 2022-09-03:
Type: Bug Report

**Describe the bug**
- OS and Version: Mac os 12.5.1
- VS Code Version: 1.71
- Embedded Tools Extension Version: 0.3.220903001
- Target embedded device:
- RTOS version:
- Other extensions you installed (and if the issue persists after disabling them):
- A clear and concise description of what the bug is.

Plugin does not detect STM32CubeIDE but ide was installed. Additionally I cannot select .cproject file - possibility for selecting this file is disabled.
[image]

benmcmorran, 2022-09-09:
@Michal123456747 this issue should now be fixed in the pre-release version of Embedded Tools. Click "Switch to Pre-Release Version" and ensure you have version v0.3.220909001 or later installed. Note that `code` must exist on the `PATH` environment variable for conversion to be successful. If it's not already, [follow these steps to add it](https://code.visualstudio.com/docs/setup/mac#_launching-from-the-command-line).

[image]

Michal123456747, 2022-09-09:
Now plugin works correctly. Really thank you for help. Unfortunately I get error, when I'm starting debugging. I will open new issue.

closed 2022-09-09 by Michal123456747