microsoft/vscode-embedded-tools #21: Unable to start debugging. No process is associated with this object when starts debugging

Michal123456747, 2022-09-09:
Type: Bug Report

OS and Version: Mac os 12.5.1
VS Code Version: 1.71
Embedded Tools Extension Version:v0.3.220909001
Target embedded device: Nucleo F446RE

Description:
When I launch program, I get this error. Next I get information: "ERROR: Unable to start debugging. No process is associated with this object." I'm so confused, because I can open arm-none-eabi-gdb file.

[image]

robotdad, 2022-09-21:
I don't think we've been able to replicate this yet. In discussions we suspect it is a signing issue. See section 5.1 of [ST's STM32CubeIDE installation manual](https://www.st.com/resource/en/user_manual/um2563-stm32cubeide-installation-guide-stmicroelectronics.pdf). The guide walks you through trusting the STM32CubeIDE.app either via macOS system preferences or the command line. Looking at the error message I would hope that is enough as gdb is within that path. This is the cli command from the docs.
```
xattr -c /Applications/STM32CubeIDE.app
```

Michal123456747, 2022-09-22:
@robotdad thanks for your reply. I executed this command and now everything works correctly. But when I start debugging without plugged in stm microcontroller I get the same information as above. Please change it to be correct when board is not connected to pc.