microsoft/vscode-embedded-tools #23: RTOS Viewer does not display Threads when used with Cortex-Debug

joel-jetcharge, 2022-09-26:
Type: Bug Report

**Describe the bug**
- OS and Version: macOS 12.5.1
- VS Code Version: 1.71.2
- Embedded Tools Extension Version: 0.3.0
- Target embedded device: STM32F07ZG
- RTOS version: Azure RTOS 6.1.12
- Other extensions you installed (and if the issue persists after disabling them): Cortex-Debug 1.6.4

When debugging with Cortex-Debug the RTOS Viewer does not load the 'Threads' details. Some communication is working, as I can see Cortex-Debug is receiving the thread names to display in the Call Stack, however the RTOS Viewer fails to load with "An error occurred while loading details. Make sure the debugger is paused and try again."

[image]

I'll note that when I view other elements such as Mutexes/Queues/Timers/etc the view loads correctly and I can see the expected output. Switching the view back to Threads once again gives the same error.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a Cortex-Debug session.
2. Run the application long enough for the RTOS to initialise the threads.
3. Hit a breakpoint or click Pause.
4. Click Embedded Tools: RTOS Viewer tab.
5. See error.

**Expected behavior**
RTOS Viewer successfully loads the Threads view.

**Code sample and logs**
- Logs from the Embedded Tools Output Window: [embedded-tools.log](https://github.com/microsoft/vscode-embedded-tools/files/9644245/embedded-tools.log)
- `launch.json` settings:
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Build & Debug Elf",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceFolder}/build/debug/<REDACTED_FILENAME>.elf",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "interface": "swd",
            "device": "STM32F407ZG",
            "preLaunchTask": "buildScons",
            "rttConfig": {
              "enabled": true,
              "address": "auto",
              "decoders": [
                  {
                      "port": 0,
                      "type": "console"
                  }
              ]
            },
            "svdPath": "${workspaceFolder}/library/mcal/bsp/STM32F4xxxx/DeviceDefinitions/STM32F407.svd"
        }
    ]
}
```

Any help is greatly appreciated :)

benmcmorran, 2022-09-28:
Thanks for the report! I was able to reproduce this issue and opened a PR with a fix to the Cortex-Debug project. https://github.com/Marus/cortex-debug/pull/750

SeanMollet, 2022-10-07:
Confirming that updating cortex-debug to master (after his merge of your PR) fixes this.
MacOS 12.6

benmcmorran, 2022-10-07:
Thanks for confirming!