microsoft/vscode-embedded-tools #24: Building Throwing Error

RaysoftTechnologiesInc, 2022-09-27:
I am trying to import my ST project into VS Code, I have followed all the steps in the blog I saw on Facebook, however I am getting that error when I run the debug command
[image]
I am not entirely sure what the issue is, but I am hoping that you can give me some pointers. My code is building successfully in CubeIDE
[cproject.zip](https://github.com/microsoft/vscode-embedded-tools/files/9652320/cproject.zip)
[st-project.zip](https://github.com/microsoft/vscode-embedded-tools/files/9652321/st-project.zip)
I have attached my cmake file and also my cproject file.

benmcmorran, 2022-09-28:
Thanks for the report! Based on your uploads, I was able to reproduce this issue with a simple project. A fix should be available on our pre-release channel within the next few days, but in the meantime you should be able to work around the issue by removing double quotes from absolute include paths within STM32CubeIDE.

[image]

RaysoftTechnologiesInc, 2022-09-30:
Hi Ben, 

Thank you for the reply, I had to reset my pc because of some issues, but now I am trying to install and activate vcpkg again, and i am now getting this error.

[image]

when I run the vcpkg activate command. What maybe be the issue here?

benmcmorran, 2022-09-30:
@BillyONeal any ideas on the `vcpkg activate` issue shown above?

Once it's working, you can run VS Code in a vcpkg environment simply by running `vcpkg activate` followed by `code .` in the terminal. As a workaround, for these projects vcpkg is only used to acquire CMake and Ninja. If you already have these tools installed and available on `PATH` you can safely ignore the warning about vcpkg environments.

BillyONeal, 2022-09-30:
1. Is your `VCPKG_ROOT` set to `C:\Users\Raymond Mango\.vcpkg` ?
2. Does `C:\Users\Raymond Mango\.vcpkg\vcpkg.exe` exist?
3. If you create `C:\Users\Raymond Mango\.vcpkg\node_modules`, does that fix it?

Thanks!

RaysoftTechnologiesInc, 2022-10-03:
Hi Billy,

1) So I have et the VCPKG_ROOT to C:\Users\Raymond Mango\.vcpkg.
2) Yes the .exe file exists.
3) I tried creating it, but running the command vcpkg activate deletes th folder and gives the same error again. i am not sure where I am going wrong because I seemed to get it right the first time. I am regretting restarting my PC.

Is there a link I can follow up the steps of installing the vcpkg from scratch I want to maybe redo everything from scratch.

BillyONeal, 2022-10-03:
@benmcmorran Is there a way to get VS to run vcpkg with `--debug`?

benmcmorran, 2022-10-03:
We don't currently have a way to add additional arguments to the command line invocation, but it looks like OP was running vcpkg in a VS Code integrated terminal, not VS. Would simply running `vcpkg activate --debug` get the information you're looking for?

BillyONeal, 2022-10-04:
I hope!

RaysoftTechnologiesInc, 2022-10-04:
I have run the command  vcpkg activate --debug. this is what I got

[image]

benmcmorran, 2022-10-10:
@BillyONeal do you see anything obvious in the log above?

escherstair, 2022-10-11:
> I am trying to import my ST project into VS Code, I have followed all the steps in the blog I saw on Facebook,

@RaysoftTechnologiesInc can you post the link to the page describing how to import ST project into VS Code?
Thanks

benmcmorran, 2022-10-11:
@escherstair https://devblogs.microsoft.com/cppblog/importing-st-projects-into-visual-studio-code/

BillyONeal, 2022-10-11:
> @BillyONeal do you see anything obvious in the log above?

I do not. This line looks like it got cut off:

[image]

It looks like we tried to download vcpkg-ce, tried to ask npm to unpack it, and somehow that produced no output but did nothing and returned a success code. We then try to mark that we pulled in the right bits and that explodes because nothing got extracted... I think. But where we would expect a command line to be printed etc. of what we invoked is garbled/missing. As a result I'm not sure how to proceed without repro instructions :(

benmcmorran, 2022-10-14:
A fix for the original issue to correctly handle double quotes around absolute include paths is now available in version 0.4.0 of the Embedded Tools extension, which is available on the VS Code Marketplace. Since the original issue is resolved, I'm going to close this issue, but @RaysoftTechnologiesInc if you're still running into issues with vcpkg on the command line, consider opening an [issue on the vcpkg repo](https://github.com/microsoft/vcpkg/issues) where they should be better equipped to help out.

closed 2022-10-14 by benmcmorran