microsoft/vscode-embedded-tools #26: Import uVision project to VS Code

escherstair, 2022-10-19:
I see that MS (together with ST) did a great job creating a way to [import STCube projects into VSCode](https://devblogs.microsoft.com/cppblog/importing-st-projects-into-visual-studio-code/).
I'd like having the same feature for Keil uVision .projx projects.

Is there someone else interested in this feature?

benmcmorran, 2022-10-19:
Thanks for the feedback @escherstair! This work isn't part of our short-term plans, but I'm marking the issue as an enhancement and leaving it open so that others can upvote and comment.

escherstair, 2022-10-24:
I add here the link to [Keil Importer script](https://github.com/damogranlabs/VS-Code-STM32-IDE/blob/master/README_KEIL_IMPORTER.md) in case it could be useful as in inspiration on how to implement this feature.