microsoft/vscode-embedded-tools #27: Cmake include directories errors

Dr-Itachi, 2022-11-05:
Hi. I'm having theses errors when I import my projrct from STM32CubeIDE. I don't know what is need to be done here
[image]

benmcmorran, 2022-11-07:
Thanks for the report! It looks like this is a bug with our handling of the [`workspace_loc` variable](https://rtist.hcldoc.com/help/index.jsp?topic=%2Forg.eclipse.platform.doc.user%2Fconcepts%2Fconcepts-exttools.htm) in STM32CubeIDE projects. I'll work on a fix in the extension, but in the meantime you should be able to work around the issue by editing the paths in the `st-project.cmake` file to replace `${workspace_loc:/${ProjName}/<the rest of the path>}` with `../<the rest of the path>`. For example, the line 22 in your screenshot would become:

```
"$<$<AND:$<CONFIG:Debug>,$<COMPILE_LANGUAGE:ASM>>:${PROJECT_SOURCE_DIR}/build/../Drivers/CMSIS/Core/Include>"
```

Dr-Itachi, 2022-11-07:
> Thanks for the report! It looks like this is a bug with our handling of the [`workspace_loc` [...]

Thanks. let me try

benmcmorran, 2022-11-30:
This issue is fixed in version 0.5.0 of the extension, which is now available in the VS Code Marketplace. The fix should support all paths of the form ${workspace_loc:/${ProjName}/<the rest of the path>}, and partially support paths of the form `${workspace_loc:/<external project name>/<the rest of the path>}`. Thanks again for your feedback!

closed 2022-11-30 by benmcmorran