microsoft/vscode-embedded-tools #30: Support STM32CubeIDE excluded files from file generation

Imrekorf, 2022-12-27:
Type: Feature Request

Currently the st-project.cmake generated from an stm32cubeide .cproject does not take into account any excluded files.|
These files can be found in the .cproject under 
```xml
<sourceEntries>
<entry excluding="file1|file2|folderpath" />
</sourceEntries>
```

benmcmorran, 2022-12-27:
Thanks for the report @Imrekorf. What version of the Embedded Tools extension are you using? Support for exclusions should have been [added in version 0.5.0](https://marketplace.visualstudio.com/items/ms-vscode.vscode-embedded-tools/changelog), which was released on November 30 (see #29). If that's not working for you, are you able to attach a minimal project that reproduces the issue? That will help us debug the problem.

Imrekorf, 2022-12-28:
[ExcludedFileMin.zip](https://github.com/microsoft/vscode-embedded-tools/files/10313003/ExcludedFileMin.zip)
Hi I've added the minimal project that reproduces the problem for me. I've marked the test folder inside the CustomHAL folder as "exclude from build" for both Debug and Release. 
I'm using stm32cubeide 1.11.0

benmcmorran, 2022-12-28:
Thanks! I was able to track the issue down to a bug in our handling of excluded directories. A fix should be publicly available in the pre-release version of the extension in the next few days. I'll let you know when it's available.

Were you also seeing issues with exclusions for individual files (`<entry excluding="foo/bar.c" />`)? The issue I identified only applied to directories (`<entry excluding="foo" />`) and files exclusions seemed to already work correctly, but if you were seeing issues with individual file exclusions too, I can also take a look at that.

benmcmorran, 2022-12-29:
This issue is fixed in pre-release version `0.5.221229002` of the Embedded Tools. You can switch to the pre-release version (which updates nightly) in the VS Code Marketplace.

[image]

benmcmorran, 2023-02-09:
This change is in version 0.6.0 of the extension, which is now available in the VS Code Marketplace.

closed 2023-02-09 by benmcmorran