microsoft/vscode-embedded-tools #34: STM32 Unhandled workspace_loc variable in Touchgfx dual core projects

pkooiman, 2023-01-30:
When generating Touchgfx projects for dual core boards such as STM32H745i disco, the .cproject file for the CM7 processor contains include paths of the form

`<listOptionValue builtIn="false" value="&quot;${workspace_loc:/Drivers/BSP/Components/Common}&quot;"/>`

These end up in st-project,cmake as `"$<$<AND:$<CONFIG:Debug>,$<COMPILE_LANGUAGE:ASM>>:${PROJECT_SOURCE_DIR}/build\\${workspace_loc:\\Drivers\\BSP\\Components\\Common}>"`
which is a syntax error.

Now, it is not clear why the STM tools add the include path, it seems completely unnecessary as the .cproject also contains the include path
`"../../../Drivers/BSP/Components/Common"/>`
which is the same. I delete the offending lines from st-project.cmake and everything works, but after every modification in Touchgfx they show up again which quickly gets old ;-)

Generating the "Blank UI" example for STM32H745i disco and importing STM32CubeIDE\CM7\.cproject into VS Code reproduces the error. There is no problem with single core boards, only dual core ones.

benmcmorran, 2023-02-02:
Thanks for the report! I can reproduce this issue. We can't easily fix this in the general case, because the ST project import feature has no knowledge of which directory was the workspace root in STM32CubeIDE. That said, I am updating the importer to drop paths that contain Eclipse macros we don't know how to interpret, which should fix the syntax errors you're getting. I'll let you know when this fix is available in the public pre-release channel.

benmcmorran, 2023-02-07:
This issue is fixed in pre-release version `0.5.230207001` of the Embedded Tools. You can switch to the pre-release version (which updates nightly) in the VS Code Marketplace. Note that as discussed above, this change does not actually add full support for general `${workspace_loc}` macros, but does remove all unknown Eclipse macros.

[image]

benmcmorran, 2023-02-09:
This change is in version 0.6.0 of the extension, which is now available in the VS Code Marketplace.

closed 2023-02-09 by benmcmorran