microsoft/vscode-embedded-tools #35: How to actually use this extension?

luverolla, 2023-02-02:
I can't find a documentation or user guide on this extension anywhere.
I tried to use it with some Zephyr samples (the canonical blinky) but nothing showed up (no relative buttons on the sidebar, no associated command when pressing Ctrl+Shift+P, no information on the bottom status bar)
I'm surely missing something

---

My setup:
Windows 10
C/C++ extension installed
All Zephyr stuff (zephyr base, SDK, west tool and python packages) installed
An ST Nucleo F401RE connected with drivers installed

benmcmorran, 2023-02-02:
Thanks for trying the extension! A good place to start is probably @robotdad's [blog post](https://devblogs.microsoft.com/cppblog/vscode-embedded-development/) and the [launch configurations from the Azure RTOS getting started guides](https://github.com/azure-rtos/getting-started/blob/master/MXChip/AZ3166/.vscode/launch.json) (which are broadly the same regardless of which RTOS you use).

Specifically for viewing Zephyr RTOS objects, the RTOS viewer won't show up until you are both in a debug session in VS Code and the RTOS viewer detects that Zephyr is running (internally it looks for the `_kernel_thread_info_offsets` or `_track_list_k_mutex` symbols to indicate the presence of Zephyr). The Embedded Tools extension today doesn't dictate a specific debug setup, and it's largely up to you how you want to actually start the debug session. If your debug setup uses GDB and a gdbserver (like OpenOCD or ST-Link), you should be able to start debugging with a `launch.json` file similar to the one I linked above, but other debug adapters like those provided by Cortex-Debug or IAR's debug extension should also work.

robotdad, 2023-02-03:
The extension overview page also links to the [official docs](https://learn.microsoft.com/en-us/cpp/embedded/?view=msvc-170). I'd be very interested in discussing in more depth what your challenges are. This is something we're digging into right now, how to make it easy to on board. I happen to have one of the boards you mention, so if there is an existing project on GitHub we could use to iterate on, or a Zephyr sample to start from, I'd be happy to work through that with you.

e-sr, 2023-04-17:
Hi,
I had also some trouble in usage and i think the cause was that i was used to use the `cortex-debug` extension and its `lanuch.json` config.

Also the [getting started repo](https://github.com/azure-rtos/getting-started) make also use of `cortex-debug` extension which is misleading, i thougth that it is necessary... 

But looking at the **NXP** example I could setup the correct  `launch.config` configuration:

```json
   {
      "name": "launch",
      "type": "cppdbg",
      "request": "launch",
      "cwd": "${workspaceFolder}",
      "program": "${workspaceFolder}/${input:project}/build/${input:project}.elf",
      "MIMode": "gdb",
      "miDebuggerPath": "/opt/gcc-arm-none-eabi/bin/arm-none-eabi-gdb",
      "miDebuggerServerAddress": "localhost:3333",
      "debugServerPath": "/usr/bin/JLinkGDBServerCLExe",
      "debugServerArgs": "-if swd -speed 12000 -device MSP432E401Y -noir -nogui -nohalt -strict -port 3333  -x ${workspaceFolder}/.vscode/reset.gdb",
      "serverLaunchTimeout": 20000,
      "serverStarted": "Connected to target",
      "launchCompleteCommand": "None",
      "filterStdout": true,
      "stopAtConnect": true,
      //"preLaunchTask": "Flash",
      "svdPath": "${workspaceFolder}/msp432ecomponents/debugfiles/MSP432E401Y.svd",
      "setupCommands": [
        {
          "description": "Allow reading of peripheral registers",
          "text": "set mem inaccessible-by-default off",
          "ignoreFailures": true
        }
      ]
    }
``` 
Also  the cmake Toolchain files are also a good starting point understanding the cmake toolchain nightmare...

Hope this can help