microsoft/vscode-embedded-tools #36: vcpkg: Failed to activate environment

froha, 2023-02-10:
Hi,

we are using already vcpkg in our project.
The new version of this extension supports it, but the activation fails.

Environment variables:
HTTP_PROXY = "http://127.0.0.1:9000"
HTTPS_PROXY = "http://127.0.0.1:9000"
VCPKG_ROOT = "C:\folder1\vcpkg"

vcpkg output:

> Using vcpkg executable at "c:\Users\Abcdefghjkl.Mnopqrst\.vscode\extensions\ms-vscode.vscode-embedded-tools-0.6.0-win32-x64\assets\platform\win32-x64\vcpkg.exe".
> Using VCPKG_ROOT at "c:\Users\Abcdefghjkl.Mnopqrst\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root".
> Activating environment at "c:\folder1\folder2\folder3\folder4\vcpkg-configuration.json".
> Version of vcpkg used to create VCPKG_ROOT: "2023-01-24-8a88d63f241d391772fbde69af9cab96c3c64c75".
> Running vcpkg with arguments ["version"].
vcpkg package management program version 2023-01-24-8a88d63f241d391772fbde69af9cab96c3c64c75

See LICENSE.txt for license information.
> Bundled version of vcpkg: "2023-01-24-8a88d63f241d391772fbde69af9cab96c3c64c75".
> Running vcpkg with arguments ["activate","--project","c:\\folder1\\folder2\\folder3\\folder4","--json","C:\\Users\\ABCDEF~1.MNO\\AppData\\Local\\Temp\\5498db1ee5c2b10b3d88c1dc7d0a2aa7"].
warning: vcpkg-ce ('configure environment') is experimental and may change at any time.
read_contents("C:\Users\Abcdefghjkl.Mnopqrst\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\scripts\vcpkgTools.xml"): no such file or directory
> Failed with exit code "1".
> Failed to activate environment at "c:\folder1\folder2\folder3\folder4\vcpkg-configuration.json".

The activation failed and the enviroment variable VCPKG_ROOT is ignored.
Using VCPKG_ROOT at "c:\Users\Abcdefghjkl.Mnopqrst\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root".

And don't use the following environment variable wit a slash at the end:
HTTP_PROXY = "http://127.0.0.1:9000/"
HTTPS_PROXY = "http://127.0.0.1:9000/"
It will fail, that is also a problem with other tools/extension, because the need the slash at the end.

froha, 2023-02-10:
After deleting the folder it is working now, (now i have an other error)
I can reproduce the reported error.
If the environment variable HTTP_PROXY has trailing slash (HTTP_PROXY = "http://127.0.0.1:9000/"), then it will fail.
It doesn't help to set the HTTP_PROXY without a trailing slash.
You have to delete the vcpkg folder like @benmcmorran mention it.

I set the VCPKG_ROOT to a special folder, which is excluded from the antivirus scanner.
This reduces the compilation time by half.
So it would be nice to set the VCPKG_ROOT in the extension.
Actual it is a really long path for the arm gnu toolchain:
c:\users\abcdefghijk.lmnopqrs\appdata\roaming\code\user\globalstorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\artifacts\c44ef60a\arm.gnu.toolchain.arm.gnu.toolchain\12.2.1\lib\gcc\arm-none-eabi\12.2.1\include\stdint.h

The vcpkg activation is now working, but now i have a build error.
`[build] c:\users\abcdefghijk.lmnopqrs\appdata\roaming\code\user\globalstorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\artifacts\c44ef60a\arm.gnu.toolchain.arm.gnu.toolchain\12.2.1\lib\gcc\arm-none-eabi\12.2.1\include\stdint.h:9:16: fatal error: stdint.h: No such file or directory
[build]     9 | # include_next <stdint.h>
[build]       |                ^~~~~~~~~~
[build] compilation terminated.`

The stdint.h file is in the correct place.
The same project activated with vcpkg over the powershell is working.
Perhaps the long path is a problem for the gcc.
I will test it next week.

froha, 2023-02-16:
@benmcmorran I don't think this issue with activating the environment failed is fixed.
If the user has wrong HTTP(S)_PROXY environment variables set, the installation of vcpkg fail.
And the user must delete the vcpkg folder to get it work with the correct HTTP(S)_PROXY environment variables.
If it is not necessary do delete the vcpkg folder (after wrong environment variables), i would say it is fixed.
At least it should be documented.
Perhaps also the HTTP(S)_PROXY variables should  be documented, that a slash is not allowed at the end.
I have at least one program (needed for a Visual Studio Code extension), which needs the slash at the HTTP(S)_PROXY environment variables to get it working with the proxy.

benmcmorran, 2023-02-16:
Thanks for clarifying. Updating the issue title.

benmcmorran, 2023-02-23:
Since it sounds like the main issue here is the lack of support for trailing slashes in vcpkg proxy settings, I'm transferring this issue to the vcpkg repo.