microsoft/vscode-embedded-tools #37: vcpkg: too long file path for gcc?

froha, 2023-02-13:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 10 21H2
- VS Code Version: 1.75.1
- Embedded Tools Extension Version: 0.6.0

**To Reproduce**
Steps to reproduce the behavior:
Download your example project:
[https://github.com/azure-rtos/getting-started](https://github.com/azure-rtos/getting-started)
Open following folder:
STMicroelectronics/B-L4S5I-IOT01A
Wait for the integrated vcpkg to activate the project.
Build

**Expected behavior**
No errors

**Code sample and logs**
`[build] [1/1294] Building C object lib/threadx/CMakeFiles/threadx.dir/common/src/tx_thread_info_get.c.obj
[build] FAILED: lib/threadx/CMakeFiles/threadx.dir/common/src/tx_thread_info_get.c.obj 
[build] C:\Users\Abcdefghijk.Lmnopqrs\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\artifacts\vcpkg-ce-default\compilers.arm.gcc\2020.10.0\bin\arm-none-eabi-gcc.exe -DTX_INCLUDE_USER_DEFINE_FILE -IC:/code/getting-started/shared/lib/threadx/ports/cortex_m4/gnu/inc -Ilib/threadx/custom_inc -isystem C:/code/getting-started/shared/lib/threadx/common/inc -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -g3 -ffunction-sections -fdata-sections -fno-strict-aliasing -fno-builtin -fno-common -Wall -Wshadow -Wdouble-promotion -Werror -Wno-unused-parameter -O0 -std=gnu99 -MD -MT lib/threadx/CMakeFiles/threadx.dir/common/src/tx_thread_info_get.c.obj -MF lib\threadx\CMakeFiles\threadx.dir\common\src\tx_thread_info_get.c.obj.d -o lib/threadx/CMakeFiles/threadx.dir/common/src/tx_thread_info_get.c.obj -c C:/code/getting-started/shared/lib/threadx/common/src/tx_thread_info_get.c
[build] In file included from C:/code/getting-started/shared/lib/threadx/ports/cortex_m4/gnu/inc/tx_port.h:85,
[build]                  from C:/code/getting-started/shared/lib/threadx/common/inc/tx_api.h:121,
[build]                  from C:/code/getting-started/shared/lib/threadx/common/src/tx_thread_info_get.c:28:
[build] c:\users\Abcdefghijk.Lmnopqrs\appdata\roaming\code\user\globalstorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\artifacts\vcpkg-ce-default\compilers.arm.gcc\2020.10.0\arm-none-eabi\include\stdlib.h:10:10: fatal error: machine/ieeefp.h: No such file or directory
[build]    10 | #include <machine/ieeefp.h>
[build]       |          ^~~~~~~~~~~~~~~~~~
[build] compilation terminated.`

**Additional context**
Too long file path?
`C:\Users\Abcdefghijk.Lmnopqrs\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\artifacts\vcpkg-ce-default\compilers.arm.gcc\2020.10.0\arm-none-eabi\include\machine\ieeefp.h`
Already mention it in the issue #36 , but it should be a new issue.
Possible solution: enable to move the vcpkg root directory

benmcmorran, 2023-02-13:
Thanks for the report! To help narrow down if long paths are causing the issue, can you try [enabling long path support](https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=powershell#enable-long-paths-in-windows-10-version-1607-and-later) (if it's not already enabled) and then restarting?

```powershell
New-ItemProperty -Path "HKLM:\SYSTEM\CurrentControlSet\Control\FileSystem" -Name "LongPathsEnabled" -Value 1 -PropertyType DWORD -Force
```

I'm also working on a change to enable overriding the storage location for the vcpkg installation, which should be available on the pre-release channel within a few days.

froha, 2023-02-14:
Long path support was enbaled.
Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem\LongPathsEnabled = 1

A colleague has the same error, but the long path support was not enabled.
Enabling it and restart his pc didn't change anything.

froha, 2023-02-14:
Additinal info:
a small test with vcpkg

Working:
$Env:VCPKG_ROOT = "C:\folder0asd\folder1asdfasdf\folder2\folder3\fol4\fol5\folder6asdfas\folder7asdfasdfasdfasdfasdfas"

Not working:
$Env:VCPKG_ROOT = "C:\folder0asd\folder1asdfasdf\folder2\folder3\fol4\fol5\folder6asdfas\folder7asdfasdfasdfasdfasdfasd"

Didn't find anything about a limit for gcc, but i had not much time today.

benmcmorran, 2023-02-14:
Your most recent comment looks like this is definitely a long path issue. I'm hoping to have a change in the pre-release channel tomorrow that enables overriding the vcpkg storage location. I'll let you know once it's available for testing, and thanks for all your help diagnosing this issue!

froha, 2023-02-15:
Nice, it is working now.
Thank you!

benmcmorran, 2023-02-15:
Thanks for confirming! I'm marking this as `fixed-pending-release` since the workaround seems pretty reasonable, but we'll continue to evaluate other options internally to see if there's a better default path we can use for vcpkg storage.