microsoft/vscode-embedded-tools #39: Extension fails to bootstrap vcpkg on Windows

JonatanAntoni, 2023-02-24:
I am running VSCode on Windows with the project [Open-CMSIS-Pack/vscode-get-started](https://github.com/Open-CMSIS-Pack/vscode-get-started) containing a `vcpkg-configuration.json` in its root.

vcpkg fails with the following output:
```
> Using vcpkg executable at ".vscode\extensions\ms-vscode.vscode-embedded-tools-0.6.0-win32-x64\assets\platform\win32-x64\vcpkg.exe".
> Using VCPKG_ROOT at "AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root".
> Activating environment at "vscode-get-started\vcpkg-configuration.json".
> Version of vcpkg used to create VCPKG_ROOT: "2023-01-24-8a88d63f241d391772fbde69af9cab96c3c64c75".
> Running vcpkg with arguments ["version"].
vcpkg package management program version 2023-01-24-8a88d63f241d391772fbde69af9cab96c3c64c75

See LICENSE.txt for license information.
> Bundled version of vcpkg: "2023-01-24-8a88d63f241d391772fbde69af9cab96c3c64c75".
> Running vcpkg with arguments ["activate","--project","vscode-get-started","--json","AppData\\Local\\Temp\\d5da59d60e5fb3221061526b6de69ca4"].
warning: vcpkg-ce ('configure environment') is experimental and may change at any time.
Downloading vcpkg-artifacts bundle 2023-01-24...
Downloading https://github.com/microsoft/vcpkg-tool/releases/download/2023-01-24/vcpkg-ce.tgz
write_contents("AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\node_modules\ce-sha.txt"): no such file or directory
> Failed with exit code "1".
> Failed to activate environment at "vscode-get-started\vcpkg-configuration.json".
```

Any ideas what's going wrong? Running the same on Linux (WSL Ubuntu) works as expected.

benmcmorran, 2023-02-24:
Are you behind a proxy? We saw a similar issue in https://github.com/microsoft/vcpkg/issues/29815 (originally reported in this repo) where vcpkg itself doesn't handle proxy URLs with a trailing slash well.

You can also try deleting the vcpkg storage location (`%APPDATA%\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg`) which will force the extension start the bootstrapping process from scratch again. Finally, there's a new `vcpkg.storageLocation` setting available on the pre-release channel (see #37) that provides a way to override where the extension stores vcpkg files. However, that was added primarily as a way to work around long path issues on Windows, which doesn't look like it's the case here.

JonatanAntoni, 2023-02-27:
I am not behind a proxy. I removed the folder `%APPDATA%\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg` but still the bootstrapping fails. I removed the extension from Code, removed all folders and reinstalled the extension. I even tried to set `vcpkg.storageLocation` to some shorter path but same result.

I cannot see this file `vcpkg\root\node_modules\ce-sha.txt` on my Windows machine, not even the folder `node_modules` is there. Looking to the same folder on a Linux box (WSL Ubuntu) this file is present. If I create the folder manually, it gets purged on rerunning the `Àctivate vcpkg environment` in Code.

benmcmorran, 2023-02-27:
Thanks for checking all that. I'm not sure exactly what's going on, but can you try following these steps with more verbose logging to see if we can get any insights? Other than the logging level changes, this should be the same thing that VS Code does internally. In PowerShell, replacing `vscode-get-started` in the last line with the path to your project directory:

```pwsh
$vcpkg = "$Env:USERPROFILE\.vscode\extensions\ms-vscode.vscode-embedded-tools-0.6.0-win32-x64\assets\platform\win32-x64\vcpkg.exe"
$Env:VCPKG_ROOT = "$Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root"
rm -Recurse $Env:VCPKG_ROOT
& $vcpkg bootstrap-standalone --debug
& $vcpkg activate --project vscode-get-started --debug
```

JonatanAntoni, 2023-03-06:
Hi @benmcmorran,

Here are the results from above commands:

```powershell
$vcpkg = "$Env:USERPROFILE\.vscode\extensions\ms-vscode.vscode-embedded-tools-0.6.230304001-win32-x64\assets\platform\win32-x64\vcpkg.exe"

$Env:VCPKG_ROOT = "$Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root"

rm -Recurse $Env:VCPKG_ROOT

& $vcpkg bootstrap-standalone --debug
[DEBUG] To include the environment variables in debug output, pass --debug-env
[DEBUG] Feature flag 'binarycaching' unset
[DEBUG] Feature flag 'compilertracking' unset
[DEBUG] Feature flag 'registries' unset
[DEBUG] Feature flag 'versions' unset
Downloading standalone bundle 2023-01-24.
Downloading https://github.com/microsoft/vcpkg-tool/releases/download/2023-01-24/vcpkg-standalone-bundle.tar.gz
[DEBUG] Found path: C:\Windows\system32\tar.exe
[DEBUG] Found path: c:\tools\git\usr\bin\tar.exe
[DEBUG] CreateProcessW("C:\Windows\system32\tar.exe" xzf "$Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\vcpkg-standalone-bundle.tar.gz")
[DEBUG] cmd_execute() returned 0 after 135361 us
[DEBUG] D:\a\_work\1\s\src\vcpkg\commands.bootstrap-standalone.cpp(51):
[DEBUG] Time in subprocesses: 135361 us
[DEBUG] Time in parsing JSON: 0 us
[DEBUG] Time in JSON reader: 0 us
[DEBUG] Time in filesystem: 4076 us
[DEBUG] Time in loading ports: 0 us
[DEBUG] Exiting after 1.235 s (1222472 us)

& $vcpkg activate --debug
[DEBUG] To include the environment variables in debug output, pass --debug-env
[DEBUG] Feature flag 'binarycaching' unset
[DEBUG] Feature flag 'compilertracking' unset
[DEBUG] Feature flag 'registries' unset
[DEBUG] Feature flag 'versions' unset
[DEBUG] Using scripts-root: $Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\scripts
[DEBUG] Bundle config: readonly=0, usegitregistry=1, embeddedsha=76a79d990dcb2e30e964322cda912db3fe395c33
[DEBUG] Using builtin-ports: $Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\ports
[DEBUG] Using installed-root: $Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\installed
[DEBUG] Using buildtrees-root: $Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\buildtrees
[DEBUG] Using packages-root: $Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\packages
[DEBUG] Using vcpkg-root: $Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root
[DEBUG] Using scripts-root: $Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\scripts
[DEBUG] Using builtin-registry: $Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\versions
[DEBUG] Using downloads-root: $Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads
warning: vcpkg-ce ('configure environment') is experimental and may change at any time.
A suitable version of node was not found (required v16.15.1) Downloading portable node 16.15.1...
Downloading node...
https://nodejs.org/dist/v16.15.1/node-v16.15.1-win-x64.7z->$Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\node-v16.15.1-win-x64.7z
Downloading https://nodejs.org/dist/v16.15.1/node-v16.15.1-win-x64.7z
Extracting node...
[DEBUG] Found path: C:\tools\cmake\bin\cmake.exe
[DEBUG] CreateProcessW("C:\tools\cmake\bin\cmake.exe" --version)
[DEBUG] ReadFile() finished with GetLastError(): 109
[DEBUG] 1000: cmd_execute_and_stream_data() returned 0 after   381876 us
[DEBUG] CreateProcessW("C:\tools\cmake\bin\cmake.exe" -E tar xzf "$Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\node-v16.15.1-win-x64.7z")
[DEBUG] cmd_execute() returned 0 after 4779472 us
[DEBUG] CreateProcessW("$Env:APPDATA\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\tools\node-16.15.1-windows\node-v16.15.1-win-x64\node.exe" --version)
[DEBUG] ReadFile() finished with GetLastError(): 109
[DEBUG] 1001: cmd_execute_and_stream_data() returned 0 after  2697113 us
Downloading vcpkg-artifacts bundle 2023-01-24...
Downloading https://github.com/microsoft/vcpkg-tool/releases/download/2023-01-24/vcpkg-ce.tgz
[DEBUG] CreateProcessW("$Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\tools\node-16.15.1-windows\node-v16.15.1-win-x64\node.exe" "$Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\tools\node-16.15.1-windows\node-v16.15.1-win-x64\node_modules\npm\bin\npm-cli.js" --force install --no-save --no-lockfile --scripts-prepend-node-path=true --silent "$Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\vcpkg-ce-2023-01-24.tgz")
[DEBUG] cmd_execute() returned 0 after 34810405 us
write_contents("$Env:APPDATA\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\node_modules\ce-sha.txt"): no such file or directory
[DEBUG] D:\a\_work\1\s\src\vcpkg\configure-environment.cpp(131):
[DEBUG] Time in subprocesses: 42668866 us
[DEBUG] Time in parsing JSON: 22 us
[DEBUG] Time in JSON reader: 0 us
[DEBUG] Time in filesystem: 17259 us
[DEBUG] Time in loading ports: 0 us
[DEBUG] Exiting after 45.16 s (45132863 us)
```

Comparing the files between Windows and Linux environments show that on Windows there is no folder `node_modules` but on Linux it exists.

benmcmorran, 2023-03-06:
Thanks. Given that the issue can reproduce on the command line, and it doesn't appear to be a long path issue based on your earlier testing of the `vcpkg.storageLocation` setting, I'm transferring the issue to the vcpkg repo. Those folks should be more familiar with the download mechanism that vcpkg uses internally and may have better insight into what would cause `node_modules` to be missing.