microsoft/vscode-embedded-tools #42: vcpkg and arm-none-eabi-gcc C++ build issue

Sn1cket, 2023-03-11:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 10 22H2 (Build 19045.2604)
- VS Code Version: 1.76.1
- Embedded Tools Extension Version: 0.6.0
- Target embedded device: MXChip_AZ3166
- Other extensions you installed (and if the issue persists after disabling them):
  * ms-vscode-remote.vscode-remote-extensionpack
  * mhutchie.git-graph
  * eamodio.gitlens
  * vscode-icons-team.vscode-icons
  * oderwat.indent-rainbow
  * ms-vscode.cpptools-extension-pack
  * twxs.cmake
  * redhat.vscode-yaml
  * jeff-hykin.better-cpp-syntax
The issue persists after disabling the extensions.
- A clear and concise description of what the bug is:

I followed this article to bootstrap my dev environment with vcpkg: https://devblogs.microsoft.com/cppblog/vcpkg-artifacts
First I tried to build the example project all in VS Code with the vcpkg environment activated like described in the article and this worked fine.
Then I changed the extension of main.c to main.cpp and I made a few C++ compatibility changes to the project. You can find the changes in my fork of the getting-started repo (https://github.com/Sn1cket/getting-started/tree/build_with_cpp)
In this state I am able to build the project when I run the following commands in a terminal:
````
vpckg activate
cmake --preset arm-gcc-cortex-m4
cmake --build --preset arm-gcc-cortex-m4
````

The problem starts when I try to do the same in VS Code. I am getting the following error when compiling the main.cpp file:
````
[build] C:\Users\<User>\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\artifacts\vcpkg-ce-default\compilers.arm.gcc\2020.10.0\bin\arm-none-eabi-g++.exe -DJSMN_HEADER -DNX_INCLUDE_USER_DEFINE_FILE -DSTM32F4 -DSTM32F412Rx -DTX_INCLUDE_USER_DEFINE_FILE -DUSE_HAL_DRIVER -IE:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/app/. -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/threadx/ports/cortex_m4/gnu/inc -IE:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/build/lib/threadx/custom_inc -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/ports/cortex_m4/gnu/inc -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/common/inc -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/auto_ip -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/azure_iot -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/BSD -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/cloud -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/dhcp -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/dns -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/ftp -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/http -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/lwm2m -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/mdns -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/mqtt -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/nat -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/pop3 -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/ppp -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/pppoe -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/ptp -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/smtp -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/snmp -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/sntp -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/telnet -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/tftp -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/web -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/websocket -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/crypto_libraries/inc -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/nx_secure/inc -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/nx_secure/ports -IE:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/build/lib/netxduo/custom_inc -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/azure_iot/azure-sdk-for-c/sdk/inc -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/azure_iot/azure-sdk-for-c/sdk/tests/core/inc -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/azure_iot/azure_iot_security_module -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/azure_iot/azure_iot_security_module/inc -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/azure_iot/azure_iot_security_module/inc/configs/RTOS_BASE -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/netxduo/addons/azure_iot/azure_iot_security_module/iot-security-module-core/inc -IE:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/lib/stm32cubef4/Drivers/CMSIS/Device/ST/STM32F4xx/Include -IE:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/lib/stm32cubef4/Drivers/CMSIS/Include -IE:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/lib/stm32cubef4/Drivers/STM32F4xx_HAL_Driver/Inc -IE:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/lib/stm32cubef4/config -IE:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/lib/mxchip_bsp/stm_sensor/Inc -IE:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/lib/mxchip_bsp/ssd1306 -IE:/dev/projects/vcpkg-playground/getting-started/shared/src/. -IE:/dev/projects/vcpkg-playground/getting-started/shared/src/azure_iot_mqtt -IE:/dev/projects/vcpkg-playground/getting-started/shared/lib/jsmn/src -isystem E:/dev/projects/vcpkg-playground/getting-started/shared/lib/threadx/common/inc -isystem E:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/lib/wiced_sdk/inc -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -g3 -ffunction-sections -fdata-sections -fno-strict-aliasing -fno-builtin -fno-common -Wall -Wshadow -Wdouble-promotion -Werror -Wno-unused-parameter -g -MD -MT app/CMakeFiles/mxchip_azure_iot.dir/main.cpp.obj -MF app\CMakeFiles\mxchip_azure_iot.dir\main.cpp.obj.d -o app/CMakeFiles/mxchip_azure_iot.dir/main.cpp.obj -c E:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/app/main.cpp
[build] In file included from c:\users\<User>\appdata\roaming\code\user\globalstorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\artifacts\vcpkg-ce-default\compilers.arm.gcc\2020.10.0\arm-none-eabi\include\c++\10.2.1\stdlib.h:36,
[build]                  from E:/dev/projects/vcpkg-playground/getting-started/shared/lib/threadx/ports/cortex_m4/gnu/inc/tx_port.h:85,
[build]                  from E:/dev/projects/vcpkg-playground/getting-started/shared/lib/threadx/common/inc/tx_api.h:121,
[build]                  from E:/dev/projects/vcpkg-playground/getting-started/MXChip/AZ3166/app/main.cpp:6:
[build] c:\users\david\appdata\roaming\code\user\globalstorage\ms-vscode.vscode-embedded-tools\vcpkg\root\downloads\artifacts\vcpkg-ce-default\compilers.arm.gcc\2020.10.0\arm-none-eabi\include\c++\10.2.1\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory
[build]    41 | #include <bits/c++config.h>
[build]       |          ^~~~~~~~~~~~~~~~~~
[build] compilation terminated.
````
It seems like the compiler include paths don't get resolved properly in the VS Code vcpkg environment.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone this repository https://github.com/Sn1cket/getting-started/tree/build_with_cpp
2. Open MXChip/AZ3166/AZ3166.code-workspace in VS Code
3. Activate the vcpkg environment
4. Open the terminal in VS Code
5. Run cmake --preset arm-gcc-cortex-m4 && cmake --build --preset arm-gcc-cortex-m4
6. See error

**Expected behavior**
I expect the build to finish like outside of the terminal

**Code sample and logs**
- https://github.com/Sn1cket/getting-started/tree/build_with_cpp

benmcmorran, 2023-03-14:
This looks suspiciously similar to #37. Can you try switching to the pre-release channel and using the new `vcpkg.storageLocation` setting (as described in [this comment](https://github.com/microsoft/vscode-embedded-tools/issues/37#issuecomment-1430372194)) to see if this is a long path issue? That setting should be available in on the official channel in the next few days once we ship a new official release.

Sn1cket, 2023-03-15:
I tested the version on the pre-release channel and the problem is fixed when using a different location than the default. So it seems to be a long path issue.
Thanks for your help!

benmcmorran, 2023-03-22:
The `vcpkg.storageLocation` setting is now available in the official 0.7.0 release.

closed 2023-03-22 by benmcmorran