microsoft/vscode-embedded-tools #44: VCPKG error whne loading imported STM32CudeIDE project

sidprice, 2023-03-24:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11
- VS Code Version: 1.76.2
- Embedded Tools Extension Version: 0.7.0
- Target embedded device: STM32F411RE
- A clear and concise description of what the bug is.

I created a simple test project with STM32CubeMX, no code was added to the generated project.

Then, the project was loaded using the STM32 VSCode Extension v1.0.0, the load failed with the following output:

`> Using vcpkg executable at "c:\Users\sidpr\.vscode\extensions\ms-vscode.vscode-embedded-tools-0.6.0-win32-x64\assets\platform\win32-x64\vcpkg.exe".
> Using VCPKG_ROOT at "c:\Users\sidpr\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root".
> Activating environment at "d:\DataRoot\Projects\MorningClock2\MorningClock_v3_stm32g070KBT\vcpkg-configuration.json".
> Version of vcpkg used to create VCPKG_ROOT: "2023-01-24-8a88d63f241d391772fbde69af9cab96c3c64c75".
> Running vcpkg with arguments ["version"].
vcpkg package management program version 2023-01-24-8a88d63f241d391772fbde69af9cab96c3c64c75
 
See LICENSE.txt for license information.
> Bundled version of vcpkg: "2023-01-24-8a88d63f241d391772fbde69af9cab96c3c64c75".
> Running vcpkg with arguments ["activate","--project","d:\\DataRoot\\Projects\\MorningClock2\\MorningClock_v3_stm32g070KBT","--json","C:\\Users\\sidpr\\AppData\\Local\\Temp\\683d8a9626f4c240dfe9f4dd00dd0172"].
warning: vcpkg-ce ('configure environment') is experimental and may change at any time.
read_contents("C:\Users\sidpr\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\scripts\vcpkgTools.xml"): no such file or directory
> Failed with exit code "1".
> Failed to activate environment at "d:\DataRoot\Projects\MorningClock2\MorningClock_v3_stm32g070KBT\vcpkg-configuration.json".`

benmcmorran, 2023-03-24:
Thanks for the report! I suspect the root cause of this issue is in vcpkg itself, not the Embedded Tools extension. To help us diagnose the problem, can you try adding this snippet to `settings.json`, reopen the folder, and then send us the vcpkg log?

```json
"vcpkg.diagnostics.additionalArguments": [ "--debug" ]
```

sidprice, 2023-03-24:
I added that line to settings.json and now the project appears to load. It doesn't build, however that may be a separate issue.

I don't remember where the vcpkg log is kept?

benmcmorran, 2023-03-24:
Sorry, I should have been clearer. The "log" I was referring to is just the vcpkg output pane (the same source as the log in your original post). You can get to it by clicking the vcpkg item in the statusbar.

sidprice, 2023-03-24:
`
> Using vcpkg executable at "c:\Users\sidpr\.vscode\extensions\ms-vscode.vscode-embedded-tools-0.7.0-win32-x64\assets\platform\win32-x64\vcpkg.exe".
> Using VCPKG_ROOT at "c:\Users\sidpr\AppData\Roaming\Code\User\globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root".
> Activating environment at "d:\DataRoot\Projects\IO Technologies\IHP Controller\vcpkg-configuration.json".
> Version of vcpkg used to create VCPKG_ROOT: "2023-03-14-105f7f7e8a5ea10dfac9874ff9d1a6bacba5f454".
> Running vcpkg with arguments ["version"].
vcpkg package management program version 2023-03-14-105f7f7e8a5ea10dfac9874ff9d1a6bacba5f454

See LICENSE.txt for license information.
> Bundled version of vcpkg: "2023-03-14-105f7f7e8a5ea10dfac9874ff9d1a6bacba5f454".
> Running vcpkg with arguments ["activate","--project","d:\\DataRoot\\Projects\\IO Technologies\\IHP Controller","--json","C:\\Users\\sidpr\\AppData\\Local\\Temp\\55a0c2706f27594fa5de1c2269cda575"].
warning: vcpkg-artifacts is experimental and may change at any time.
Artifact                          Version Status    Dependency Summary                                             
microsoft:tools/kitware/cmake     3.20.1  installed            Kitware's cmake tool                                
microsoft:tools/ninja-build/ninja 1.10.2  installed            Ninja is a small build system with a focus on speed.

[1/2] microsoft:tools/kitware/cmake already installed.
[2/2] microsoft:tools/ninja-build/ninja already installed.
> Activated environment at "d:\DataRoot\Projects\IO Technologies\IHP Controller\vcpkg-configuration.json".
`

benmcmorran, 2023-03-24:
Thanks. Given that it looks like you're not experiencing the issue any more I don't think we can do much more right now. Let me know if it happens again. This is potentially related to https://github.com/microsoft/vcpkg/issues/30051, but it's difficult to tell for sure.

CraigCAKC, 2024-09-20:
I am seeing what looks like this same issue running vscode on Ubuntu 22.04. The differences are: 
- Ubuntu 22.04 
- Embedded Tools extension: 0.8.240918001 (pre-release) 
- project created in STM32CubeIDE for STM32G473QETx with no added code 
- first ran into issues when attempting `Clean Reconfigure All Projects` after importing CubeIDE project
- BTW: the initial install of Ubuntu 22.04 did not include `curl` (`sudo apt install curl` seemed to get rid of initial error)

As suggested for previously reported error I added the line:
`"vcpkg.diagnostics.additionalArguments": [ "--debug" ]`
to the settings.json file. Following is the output (accessed by clicking red vcpkg box): 

```
> Using vcpkg executable at "/home/craigk/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.8.240918001-linux-x64/assets/platform/linux-x64/vcpkg-glibc.app".
> Using VCPKG_ROOT at "/home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root".
> Activating environment at "/home/craigk/wd/projCAKC/generic-G473QET/vcpkg-configuration.json".
> Version of vcpkg used to create VCPKG_ROOT: "2024-08-01-fd884a0d390d12783076341bd43d77c3a6a15658".
> Running vcpkg with arguments ["version","--debug"].
[DEBUG] To include the environment variables in debug output, pass --debug-env
[DEBUG] Trying to load bundleconfig from /home/craigk/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.8.240918001-linux-x64/assets/platform/linux-x64/vcpkg-bundle.json
[DEBUG] Failed to open: /home/craigk/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.8.240918001-linux-x64/assets/platform/linux-x64/vcpkg-bundle.json
[DEBUG] Bundle config: readonly=false, usegitregistry=false, embeddedsha=nullopt, deployment=Git, vsversion=nullopt
[DEBUG] Metrics enabled.
[DEBUG] Feature flag 'binarycaching' unset
[DEBUG] Feature flag 'compilertracking' unset
[DEBUG] Feature flag 'registries' unset
[DEBUG] Feature flag 'versions' unset
[DEBUG] Feature flag 'dependencygraph' unset
vcpkg package management program version 2024-08-01-fd884a0d390d12783076341bd43d77c3a6a15658

See LICENSE.txt for license information.
[DEBUG] /source/src/vcpkg/commands.version.cpp(34): 
[DEBUG] Time in subprocesses: 0us
[DEBUG] Time in parsing JSON: 0us
[DEBUG] Time in JSON reader: 0us
[DEBUG] Time in filesystem: 341us
[DEBUG] Time in loading ports: 0us
[DEBUG] Exiting after 1.98 ms (1096us)
> Bundled version of vcpkg: "2024-08-01-fd884a0d390d12783076341bd43d77c3a6a15658".
> Running vcpkg with arguments ["activate","--json","/tmp/bb934f9317e8180b37fa51fde1298998","--debug"].
[DEBUG] To include the environment variables in debug output, pass --debug-env
[DEBUG] Trying to load bundleconfig from /home/craigk/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.8.240918001-linux-x64/assets/platform/linux-x64/vcpkg-bundle.json
[DEBUG] Failed to open: /home/craigk/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.8.240918001-linux-x64/assets/platform/linux-x64/vcpkg-bundle.json
[DEBUG] Bundle config: readonly=false, usegitregistry=false, embeddedsha=nullopt, deployment=Git, vsversion=nullopt
[DEBUG] Metrics enabled.
[DEBUG] Feature flag 'binarycaching' unset
[DEBUG] Feature flag 'compilertracking' unset
[DEBUG] Feature flag 'registries' unset
[DEBUG] Feature flag 'versions' unset
[DEBUG] Feature flag 'dependencygraph' unset
[DEBUG] Using scripts-root: /home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/scripts
[DEBUG] Using builtin-ports: /home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/ports
[DEBUG] Using installed-root: /home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/installed
[DEBUG] Using buildtrees-root: /home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/buildtrees
[DEBUG] Using packages-root: /home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/packages
[DEBUG] Using vcpkg-root: /home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root
[DEBUG] Using builtin-registry: /home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/versions
[DEBUG] Using downloads-root: /home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads
[DEBUG] External asset downloads are allowed (x-block-origin is disabled)...
[DEBUG] Asset cache is not configured.
warning: vcpkg-artifacts is experimental and may change at any time.
[DEBUG] Failed to open: /home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/scripts/vcpkgTools.xml
read_contents("/home/craigk/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/scripts/vcpkgTools.xml"): No such file or directory
[DEBUG] /source/src/vcpkg/tools.cpp(724): 
[DEBUG] Time in subprocesses: 0us
[DEBUG] Time in parsing JSON: 0us
[DEBUG] Time in JSON reader: 0us
[DEBUG] Time in filesystem: 446us
[DEBUG] Time in loading ports: 0us
[DEBUG] Exiting after 3.93 ms (2773us)
> Failed with exit code "1".
> Failed to activate environment at "/home/craigk/wd/projCAKC/generic-G473QET/vcpkg-configuration.json".
```

CraigCAKC, 2024-09-26:
After doing a completely clean install of the vscode snap on Ubuntu 22.04, I am no longer getting this error. The clean install does not have the CMake extension