microsoft/vscode-embedded-tools #46: vcpkg error: Failed to provision vcpkg-artifacts

anderwm, 2023-03-27:
Type: Bug Report

**Describe the bug**
- OS and Version: Ubuntu 20.04
- VS Code Version: 1.76.2
- Embedded Tools Extension Version: 1.0.0
- Target embedded device: N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to load any workspace with the STM32 VS Code Extension
2. Note that vcpkg has failed with output:
```
> Using vcpkg executable at "/home/redact/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.7.230323001-linux-x64/assets/platform/linux-x64/vcpkg-glibc.app".
> Using VCPKG_ROOT at "/home/redact/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root".
> Activating environment at "/home/redact/STM32CubeIDE/workspace_1.8.0/redactredact/vcpkg-configuration.json".
> Version of vcpkg used to create VCPKG_ROOT: "2023-03-14-105f7f7e8a5ea10dfac9874ff9d1a6bacba5f454".
> Running vcpkg with arguments ["version"].
vcpkg package management program version 2023-03-14-105f7f7e8a5ea10dfac9874ff9d1a6bacba5f454

See LICENSE.txt for license information.
> Bundled version of vcpkg: "2023-03-14-105f7f7e8a5ea10dfac9874ff9d1a6bacba5f454".
> Running vcpkg with arguments ["activate","--project","/home/redact/STM32CubeIDE/workspace_1.8.0/redactredact","--json","/tmp/0cbf6b4eb2531ea7c4a1e4a69595a744"].
warning: vcpkg-artifacts is experimental and may change at any time.
Downloading vcpkg-artifacts bundle 2023-03-14...
error: Failed to provision vcpkg-artifacts.
> Failed with exit code "1".
```

**Expected behavior**
I believe it is trying to find an artifacts bundle in ~/.cache/vcpkg but in fact that directory is empty. I'm not sure why it doesn't try to download and extract the bundle.

I have tried the same thing on an Ubuntu 22.04 machine and it works as expected without this error

benmcmorran, 2023-03-27:
Thanks for the feedback. Can you ensure you are running version 0.7.0 of Embedded Tools, add this snippet to `settings.json`, and then capture the vcpkg output again? This should provide more verbose diagnostic information.

```json
"vcpkg.diagnostics.additionalArguments": [ "--debug" ]
```

anderwm, 2023-03-27:
```
> Using vcpkg executable at "/home/user_a/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.7.0-linux-x64/assets/platform/linux-x64/vcpkg-glibc.app".
> Using VCPKG_ROOT at "/home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root".
> Activating environment at "/home/user_a/STM32CubeIDE/workspace_1.8.0/project_a/vcpkg-configuration.json".
> Version of vcpkg used to create VCPKG_ROOT: "2023-03-14-105f7f7e8a5ea10dfac9874ff9d1a6bacba5f454".
> Running vcpkg with arguments ["version","--debug"].
[DEBUG] To include the environment variables in debug output, pass --debug-env
[DEBUG] Trying to load bundleconfig from /home/user_a/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.7.0-linux-x64/assets/platform/linux-x64/vcpkg-bundle.json
[DEBUG] Failed to open: /home/user_a/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.7.0-linux-x64/assets/platform/linux-x64/vcpkg-bundle.json
[DEBUG] Bundle config: readonly=false, usegitregistry=false, embeddedsha=nullopt, deployment=Git, vsversion=nullopt
[DEBUG] Metrics enabled.
[DEBUG] Feature flag 'binarycaching' unset
[DEBUG] Feature flag 'compilertracking' unset
[DEBUG] Feature flag 'registries' unset
[DEBUG] Feature flag 'versions' unset
vcpkg package management program version 2023-03-14-105f7f7e8a5ea10dfac9874ff9d1a6bacba5f454

See LICENSE.txt for license information.
[DEBUG] /mnt/vss/_work/1/s/src/vcpkg/commands.version.cpp(32): 
[DEBUG] Time in subprocesses: 0us
[DEBUG] Time in parsing JSON: 0us
[DEBUG] Time in JSON reader: 0us
[DEBUG] Time in filesystem: 102us
[DEBUG] Time in loading ports: 0us
[DEBUG] Exiting after 1.086924 ms (448us)
> Bundled version of vcpkg: "2023-03-14-105f7f7e8a5ea10dfac9874ff9d1a6bacba5f454".
> Running vcpkg with arguments ["activate","--project","/home/user_a/STM32CubeIDE/workspace_1.8.0/project_a","--json","/tmp/a4b6e267289b08b9807a7697a59e39ea","--debug"].
[DEBUG] To include the environment variables in debug output, pass --debug-env
[DEBUG] Trying to load bundleconfig from /home/user_a/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.7.0-linux-x64/assets/platform/linux-x64/vcpkg-bundle.json
[DEBUG] Failed to open: /home/user_a/.vscode/extensions/ms-vscode.vscode-embedded-tools-0.7.0-linux-x64/assets/platform/linux-x64/vcpkg-bundle.json
[DEBUG] Bundle config: readonly=false, usegitregistry=false, embeddedsha=nullopt, deployment=Git, vsversion=nullopt
[DEBUG] Metrics enabled.
[DEBUG] Feature flag 'binarycaching' unset
[DEBUG] Feature flag 'compilertracking' unset
[DEBUG] Feature flag 'registries' unset
[DEBUG] Feature flag 'versions' unset
[DEBUG] Using scripts-root: /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/scripts
[DEBUG] Using builtin-ports: /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/ports
[DEBUG] Using installed-root: /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/installed
[DEBUG] Using buildtrees-root: /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/buildtrees
[DEBUG] Using packages-root: /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/packages
[DEBUG] Using vcpkg-root: /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root
[DEBUG] Using scripts-root: /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/scripts
[DEBUG] Using builtin-registry: /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/versions
[DEBUG] Using downloads-root: /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads
warning: vcpkg-artifacts is experimental and may change at any time.
[DEBUG] 1000: popen( /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads/tools/node-16.15.1-linux/node-v16.15.1-linux-x64/bin/node --version 2>&1)
[DEBUG] 1000: cmd_execute_and_stream_data() returned 0 after     2661 us
[DEBUG] vcpkg-artifacts base path: /home/user_a/.cache/vcpkg/artifacts-2023-03-14
Downloading vcpkg-artifacts bundle 2023-03-14...
[DEBUG] 1001: popen( curl --fail -L https://github.com/microsoft/vcpkg-tool/releases/download/2023-03-14/vcpkg-ce.tgz --create-dirs --output /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads/vcpkg-ce-2023-03-14.tgz.4027009.part 2>&1)
[DEBUG] 1001: cmd_execute_and_stream_data() returned 0 after  1091962 us
[DEBUG] Trying to hash /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads/vcpkg-ce-2023-03-14.tgz.4027009.part
[DEBUG] /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads/vcpkg-ce-2023-03-14.tgz.4027009.part has hash fb2c2f568b563a88d2ebf2b29ced491e8109134529c477629e5e1dde89f52bf8a60d9b5ce8285f260cc94b7c100704128d3724bbda523edb57233dbda5a2c02a
[DEBUG] system(cd /home/user_a/.cache/vcpkg/artifacts-2023-03-14 && PATH=/home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads/tools/node-16.15.1-linux/node-v16.15.1-linux-x64/bin:/home/user_a/.local/bin:/home/user_a/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin  /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads/tools/node-16.15.1-linux/node-v16.15.1-linux-x64/bin/n [... cut ...] cpkg/root/downloads/tools/node-16.15.1-linux/node-v16.15.1-linux-x64/lib/node_modules/npm/bin/npm-cli.js --force install --no-save --no-lockfile --scripts-prepend-node-path=true --silent /home/user_a/.config/Code/User/globalStorage/ms-vscode.vscode-embedded-tools/vcpkg/root/downloads/vcpkg-ce-2023-03-14.tgz)
```
It looks like this curl fails...or puts it in the wrong directory because when it tries:
```
system(cd /home/user_a/.cache/vcpkg/artifacts-2023-03-14 ...
```
this directory doesn't exist

benmcmorran, 2023-03-27:
Thanks. The call to vcpkg from Embedded Tools looks fine to me, so I think this is an issue with vcpkg itself. I'm transferring this issue to the vcpkg repo where the maintainers should be better able to help you.