microsoft/vscode-embedded-tools #49: Placing a breakpoint while program is running throws an error

alexandru-vrn, 2023-04-10:
- OS and Version: Windows 10 Pro 22H2
- VS Code Version: 1.77.1
- Embedded Tools Extension Version: v0.7.230405001
- Target embedded device: STM32F303RE (Nucleo Board)

Placing a breakpoint somewhere in the code seems to halt the microcontroller, GDB reporting the following in the debug console tab:

> ST-LINK_gdbserver: Received Ctrl-C from the client
> ST-LINK_gdbserver: Device not halted(0), performing Ctrl-C
> ST-LINK_gdbserver: GDB session, device event: 4
> ST-LINK_gdbserver: Device halted
> ST-LINK_gdbserver: ST-LINK device status: HALT_MODE
> ST-LINK_gdbserver: GDB session, device event: 0

VSCode displays the error like this:
[image]

Resuming the program works, and the breakpoint is actually applied and reached by the program execution as expected, regardless of all of the warnings and errors displayed when it is set up.

benmcmorran, 2023-04-12:
Can you try adding this line to your `launch.json`? I made [a change to MIEngine](https://github.com/microsoft/MIEngine/pull/1351) (the underlying debug engine used by the C/C++ VS Code extenion) a while ago that should let you configure how unexpected breakpoints are handled.

```json
"unknownBreakpointHandling": "stop"
```

This won't prevent the debugger from stopping entirely (because it sounds like for some reason GDB is still stopping the program), but it should prevent it from appearing as an exception.

alexandru-vrn, 2023-04-13:
I have added the line to `launch.json` and the exception is not thrown in the way I described, but this only works once. If I add a new breakpoint then VSCode behaves like before adding the option you mentioned. If I run the program again, the exception is thrown from the first attempt of adding a new breakpoint. I have to restart VSCode and then it behaves as you describe, but only once.