microsoft/vscode-embedded-tools #52: Vcpkg fails to activate

benmcmorran, 2023-04-18:
_(reposting from feedback on Facebook)_

> Try to do it on 2 differents computer : same error , Vcpkg error. "globalStorage\ms-vscode.vscode-embedded-tools\vcpkg\root\scripts\vcpkgTools.xml"): no such file or directory

> Failed with exit code "1"."

benmcmorran, 2023-04-18:
Can you try switching to the latest pre-release version of Embedded Tools? We recently included some vcpkg updates that should make deployment less error-prone.

[image]

BR38, 2023-05-09:
I am the original writer of the Facebook post. I am just seeing this issue on github.... I tried right now and result seems the same.
[image]

benmcmorran, 2023-05-09:
(moving this issue to the vcpkg repo because it appears to be an issue with vcpkg itself, not the VS Code integration)

The error message says that `tar` is not installed. On Windows, vcpkg uses the system version of `tar` which should be included by default with Windows 10. I'd verify that `tar` is available on your system and install it if it's not.