microsoft/vscode-embedded-tools #53: Build analyser would be a great addition

Rob-le-Houe, 2023-04-21:
Hi,

I'm really enjoying the embedded tools extension for VSCode so first of all thanks for making it!

The only reason I use STM32CubeIDE now is to view the memory usage build analyser as shown in the picture. It would be great to have a tab that shows that in VSCode.

Kind regards,
Rob
[image]

froha, 2023-04-21:
Hi,

that would be great.
For programming i use the VSCode, but for some features i need the STMCubeIDE.
I also didn't find other extension for this kind of tool.

Best regards
Alex

benmcmorran, 2023-04-25:
Thanks for the feedback! I'm marking this as a feature request. We likely won't be able to implement it immediately, but I'm leaving this issue open so that others can vote and provide input.