microsoft/vscode-embedded-tools #54: Could not find an STM32CubeIDE installation

lorylia, 2023-04-29:
I have STM32CubeIDE installed, and it is the default path used, but the extension still does not find STM32CubeIDE.
[image]

benmcmorran, 2023-05-01:
As a basic check for a working STM32CubeIDE installation, the extension checks for GDB on these paths. Can you verify that your installation of GDB (provided as part of STM32CubeIDE) exists on one of them?

```
/ST/STM32CubeCLT/STM32CubeCLT/GNU-tools-for-STM32/bin/arm-none-eabi-gdb.exe
/ST/STM32CubeIDE_*/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.*/tools/bin/arm-none-eabi-gdb.exe
/opt/st/stm32cubeide_*/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.*/tools/bin/arm-none-eabi-gdb
/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.*/tools/bin/arm-none-eabi-gdb
```

benmcmorran, 2023-05-03:
Thanks, that path looks right to me. The other thing I can think of that might be causing issues is if VS Code is installed to a drive letter other than `C:` or if the project you're opening is on a drive letter other than `C:`. Can you check which drive letter VS Code is installed on and which drive letter the project you're opening is on? Not searching all drive letters is a bug, but as a workaround it should work to ensure everything is on the same drive letter.

ttrinidad, 2023-08-15:
It's silly to hard code a path like this make it so that we can input it please, if the extension wasn't closed source I would have already submitted a patch, like I patched my own local extension source.

harrybeadle-radpropulsion, 2024-06-12:
> As a basic check for a working STM32CubeIDE installation, the extension checks for GDB on these paths. Can you verify that your installation of GDB (provided as part of STM32CubeIDE) exists on one [...]

This list doesn't include the default install paths using ST's Linux installers for wither CLT or the IDE.