microsoft/vscode-embedded-tools #66: STLink GDB Server discovery does not find CLT's STLink Server.

schilkp, 2023-09-12:
Type: Bug Report

**Describe the bug**
- OS and Version: Manjaro 23.0.1
- VS Code Version: 1.82.0
- Embedded Tools Extension Version: v0.7.0
- Target embedded device: STM32L476
- RTOS version: NA

**To Reproduce**
- Install CubeCLT at the (default) location (`/opt/st/stm32cubeclt`), This will place the stlink gdbserver executeable at: `/opt/st/stm32cubeclt/STLink-gdb-server/bin/ST-LINK_gdbserver`
- Execute the following task: 
```
{
  "version": "2.0.0",
  "tasks": [
    {
      "label": "Test",
      "type": "shell",
      "command": "echo ${command:vscode-embedded.st.gdbserver}",
      "problemMatcher": []
    }
  ]
}
```

**Expected behavior**
The path of the st link gdb server is echoed into the terminal.

**Actual behaviour**
Error: Could not find the path to "ST-Link_gdbserver". Ensure STM32CubeIDE is installed.

**Additional context**
The debug message would indicate that vscode-embedded-tools is looking for the CubeIDE STLink gdbserver, as opposed to the 
CLT's. CubeIDE is also installed in the default location (`/opt/st/stm32cubeide_1.13.1/`), with it's gdb server at  `/opt/st/stm32cubeide_1.13.1/plugins/com.st.stm32cube.ide.mcu.externaltools.stlink-gdb-server.linux64_2.1.0.202305091550/tools/bin/ST-LINK_gdbserver`. According to previous changelogs this plugin should now be prioritizing the CLT version. Therefor the CubeIDE discovery is also likely broken. Furthermore, since CLT should be prioritized (according to your previous changelogs), the error message that CubeIDE should be installed is possibly wrong and misleading.

trembel, 2024-01-18:
I can confirm this.

funder7, 2024-11-03:
Same problem here, just configured the extension today on Linux.

Dannykcw, 2025-01-12:
Looking at cmake\gcc-arm-none-eabi.cmake, the file begins with the following lines:
```
set(WINDOWS_ST_CLT_PATH "C:/ST/STM32CubeCLT/STM32CubeCLT/GNU-tools-for-STM32/bin/")
set(MAC_ST_CLT_PATH "/opt/ST/STM32CubeCLT/GNU-tools-for-STM32/bin/")
```
The corresponding version for Linux is absent, which means the Linux CLT installation is never packed up. This then follows on to the following check for cubeide installation instead:
```
else()
    # Try to find an STM32CubeIDE installation to use for the toolchain.
    file(GLOB TOOLCHAIN_DIRECTORIES
        "C:/ST/STM32CubeIDE_*/STM32CubeIDE/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.*/tools/bin/"
        "/opt/st/stm32cubeide_*/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.*/tools/bin/"
        "/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.*/tools/bin/"
```
I checked with a task.json task that shows that the path of the cubeide’s gdb server is not resolved correctly on Linux. Replacing the Linux path with a hardcoded path seems to resolve the issue. However, this seems very ad hoc and very much an annoyance.