microsoft/vscode-embedded-tools #67: Pre-Release: Failed to activate environment - unexpected argument

froha, 2023-09-22:
Type: Bug Report

**Describe the bug**
- OS and Version: Win10 22H2
- VS Code Version: 1.82.2
- Embedded Tools Extension Version: v0.7.230919001 (Pre-Release)
- vcpkg cannot activate the environment

**To Reproduce**
Steps to reproduce the behavior:
1. start vscode with project folder
2. look at the red marked vcpkg at the bottom
3. open vcpkg log
4. See the vcpkg log

**Expected behavior**
Activated environment

**Code sample and logs**
`
> Using vcpkg executable at "c:\Users\xyz\.vscode\extensions\ms-vscode.vscode-embedded-tools-0.7.230919001-win32-x64\assets\platform\win32-x64\vcpkg.exe".
> Using VCPKG_ROOT at "C:\xyz\vcpkgvsc\root".
> Activating environment at "c:\xxx\xxx\xyz\xyz\vcpkg-configuration.json".
> Version of vcpkg used to create VCPKG_ROOT: "2023-09-15-ac02a9f660977426b8ec6392919fbb1d51b10998".
> Running vcpkg with arguments ["version"].
vcpkg package management program version 2023-09-15-ac02a9f660977426b8ec6392919fbb1d51b10998

See LICENSE.txt for license information.
> Bundled version of vcpkg: "2023-09-15-ac02a9f660977426b8ec6392919fbb1d51b10998".
> Running vcpkg with arguments ["activate","--project","c:\\xyz\\xyz\\xyz\\xyz","--json","C:\\Users\\xyz\\AppData\\Local\\Temp\\f26d122252ab945902f4013cbea112c4"].
error: unexpected switch: --project
error: the command 'activate' does not accept any additional arguments
error: unexpected argument: c:\xyz\xyz\xyz\xyz
vcpkg activate
Options:
  --all-languages        Acquires all language files when acquiring artifacts
  --arm                  Forces host detection to ARM when acquiring artifacts
  --arm64                Forces host detection to ARM64 when acquiring artifacts
  --x-asset-sources=...  Asset caching sources. See 'vcpkg help assetcaching'
  --binarysource=...     Binary caching sources. See 'vcpkg help binarycaching'
  --x-buildtrees-root=...
                         Buildtrees directory (experimental)
  --downloads-root=...   Downloads directory (default: %VCPKG_DOWNLOADS%)
  --force                Forces reacquire if an artifact is already acquired
  --freebsd              Forces host detection to FreeBSD when acquiring artifacts
  --host-triplet=...     Host triplet. See 'vcpkg help triplet' (default:
                         %VCPKG_DEFAULT_HOST_TRIPLET%)
  --x-install-root=...   Installed directory (experimental)
  --json=...             Full path to JSON file where environment variables and other properties are
                         recorded
  --linux                Forces host detection to Linux when acquiring artifacts
  --msbuild-props=...    Full path to the file in which MSBuild properties will be written
  --osx                  Forces host detection to MacOS when acquiring artifacts
  --overlay-ports=...    Directories of overlay ports (also: %VCPKG_OVERLAY_PORTS%)
  --overlay-triplets=... Directories of overlay triplets (also: %VCPKG_OVERLAY_TRIPLETS%)
  --x-packages-root=...  Packages directory (experimental)
  --target:arm           Sets target detection to ARM when acquiring artifacts
  --target:arm64         Sets target detection to ARM64 when acquiring artifacts
  --target:x64           Sets target detection to x64 when acquiring artifacts
  --target:x86           Sets target to x86 when acquiring artifacts
  --triplet=...          Target triplet. See 'vcpkg help triplet' (default: %VCPKG_DEFAULT_TRIPLET%)
  --vcpkg-root=...       The vcpkg root directory (default: %VCPKG_ROOT%)
  --windows              Forces host detection to Windows when acquiring artifacts
  --x64                  Forces host detection to x64 when acquiring artifacts
  --x86                  Forces host detection to x86 when acquiring artifacts

> Failed with exit code "1".
> Failed to activate environment at "c:\xyz\xyz\xyz\xyz\vcpkg-configuration.json".
`

benmcmorran, 2023-10-03:
@BillyONeal Did vcpkg intentionally remove the `--project` option from `vcpkg activate`?

BillyONeal, 2023-10-03:
Almost certainly broken by https://github.com/microsoft/vcpkg-tool/pull/1165 -- I didn't realize someone could manually say `--project`. Is that supposed to point at the `vcpkg-configuration.json` or the directory containing it?

BillyONeal, 2023-10-04:
@benmcmorran did some testing and says changing the extension to set CWD instead seems to work...

benmcmorran, 2023-10-05:
@froha This should now be fixed in the pre-release channel (v0.7.231005001 or later). Let me know if you're still seeing any issues.

froha, 2023-10-05:
It works! Thank you!

closed 2023-10-05 by froha

BillyONeal, 2023-10-05:
Thanks again @benmcmorran !