microsoft/vscode-embedded-tools #76: RTOS view thread name only shows a single character

cosmikwolf, 2024-09-13:
RTOS view thread name only shows a single character

Type: Bug Report

**Describe the bug**
- OS and Version:
 ` MacOS Sonoma 14.6.1`
- VS Code Version:
  ```
  Version: 1.93.1
  Commit: 38c31bc77e0dd6ae88a4e9cc93428cc27a56ba40
  Date: 2024-09-11T17:20:05.685Z (2 days ago)
  Electron: 30.4.0
  ElectronBuildId: 10073054
  Chromium: 124.0.6367.243
  Node.js: 20.15.1
  V8: 12.4.254.20-electron.0
  OS: Darwin arm64 23.6.0
  ```
- Embedded Tools Extension Version:
  `v0.7.0 and  v0.7.240913001` 
- Target embedded device:
  `stm32h563 using openocd over GDB`
- RTOS version:
  `threadx v6.4.1`
- Other extensions you installed (and if the issue persists after disabling them):
  `Cortex Debug v1.12.1`

- A clear and concise description of what the bug is.
  When debugging over GDB using cortex-debug, I cannot see the thread names, I just see a single character:

**To Reproduce**
- Steps to reproduce the behavior:
- Start a Cortex-Debug session.
- Run the application long enough for the RTOS to initialise the threads.
- Hit a breakpoint or click Pause.
- Click Embedded Tools: RTOS Viewer tab.
- See error.

**Expected behavior**
I expect to see the thread names. they are visible when debugging with Segger Ozone

**Code sample and logs**
- Code sample
- `launch.json`
  ```
  {
    "configurations": [
      {
        "type": "cortex-debug",
        "name": "006_pendant_board_bringup",
        "request": "attach",
        "servertype": "external",
        "gdbTarget": "localhost:3333",
        "cwd": "${workspaceFolder}",
        "executable": "${workspaceFolder}/build/projects/006_pendant_board_bringup/006_pendant_board_bringup.elf",
        "device": "STM32H563ZITx",
        "rtos": "Azure",
        "runToEntryPoint": "main",
        // "showDevDebugOutput": "raw",
        "svdPath": "/opt/ST/STM32CubeCLT_1.15.1/STMicroelectronics_CMSIS_SVD/STM32H563.svd",
        "postAttachCommands": [
          "load",
          "thbreak HardFault_Handler"
        ]
      }
    ]
  }
  ```

**Screenshots**
  [image]

**Additional context**
  seems like this might be related to https://github.com/microsoft/vscode-embedded-tools/issues/23 ?

SeanMollet, 2024-09-17:
I'm getting this too. Have to start naming my threads a-z. :)

SeanMollet, 2024-09-17:
I remembered a similar issue I'd encountered before debugging C/C++ with vscode. If the GDB isn't built with python support, pretty printing can't be enabled. I believe pretty printing is required in order for GDB to parse the entirety of C-strings. 

By default arm-none-eabi-gcc's gdb isn't built with python support.

I'm going to dig around, I've got another gdb somewhere here that supports it.

SeanMollet, 2024-09-17:
Yep.. that was it. Tried with a multiarch-gdb and it worked. To fix it without having to build your own gdb, Install the arm-none-eabi-gdb from brew, overwrite the link from the one included with the arm cask and you'll be good to go.

https://formulae.brew.sh/formula/arm-none-eabi-gdb

[image]

SeanMollet, 2024-09-17:
You can check if your gdb has python with this command:
arm-none-eabi-gdb --ex "python import os" --batch 2>&1 | grep Python

If it prints nothing, you have python and it's working. Else, it will print an error.

SeanMollet, 2024-10-09:
@rokobertcm Make sure you're using that gdb. In launch.json:

            "gdbPath": "/opt/homebrew/bin/arm-none-eabi-gdb"

rokobertcm, 2024-10-09:
Yes it works. My mistake.
Thanks!

Best regards,
Robert Konc