microsoft/vscode-embedded-tools #77: Unable to start debugging. Debug server process failed to initialize.

susano254, 2025-01-01:
Type: Bug Report

**Describe the bug**
- OS and Version: Windows 11
- VS Code Version: 1.96
- Embedded Tools Extension Version: 
- Target embedded device:
- RTOS version: 0.8.0
- Other extensions you installed (and if the issue persists after disabling them): STM32 VSCODE EXTENSION

I'm having an "Unable to start Debuggin. Debug server process failed to initialize when trying to debug using the marc goodner blog 
I confirmed it's not a hardware issue from launching the debug session from stm32cube ide itself and it works correctly and I can see the following in the debug console of vscode
 ` 
 --> E (output): {"type":"event","event":"output","body":{"category":"console","output":"
1: (155) ->        SWD Debug                  : Enabled\r\n"},"seq":28}
1: (155) ->        SWD Debug                  : Enabled

--> E (output): {"type":"event","event":"output","body":{"category":"console","output":"
1: (155) ->\r\n"},"seq":30}
1: (155) ->

--> E (output): {"type":"event","event":"output","body":{"category":"console","output":"
1: (185) ->Waiting for debugger connection...\r\n"},"seq":32}
1: (185) ->Waiting for debugger connection...

--> E (output): {"type":"event","event":"output","body":{"category":"console","output":"
1: (10139) <-logout\r\n"},"seq":34}
1: (10139) <-logout

--> E (output): {"type":"event","event":"output","body":{"category":"console","output":"
1: (10149) Send Event AD7MessageEvent\r\n"},"seq":36}
1: (10149) Send Event AD7MessageEvent

--> E (output): {"type":"event","event":"output","body":{"category":"telemetry","output":"VS/Diagnostics/Debugger/Launch","data":{"VS.Diagnostics.Debugger.ImplementationName":"Microsoft.MIDebugEngine","VS.Diagnostics.Debugger.EngineVersion":"17.12.21003.1","VS.Diagnostics.Debugger.HostVersion":"17.12.21003.1","VS.Diagnostics.Debugger.AdapterId":"cppdbg","VS.Diagnostics.Debugger.Launch.ErrorCode":1005,"VS.Diagnostics.Debugger.Launch.IsError":true}},"seq":38}

--> R (launch-2): {"type":"response","request_seq":2,"success":false,"command":"launch","message":"Unable to start debugging. Debug server process failed to initialize.","body":{"error":{"id":1005,"format":"Unable to start debugging. Debug server process failed to initialize."}},"seq":40}
 `

**Expected behavior**

Debugger connected message in the console and a succesful debug session launched in vscode

**Code sample and logs**
- `launch.json`: 
`{
  "version": "0.2.0",
  "configurations": [
    {
      "name": "Launch",
      "type": "cppdbg",
      "request": "launch",
      "cwd": "${workspaceFolder}",
      "program": "${command:cmake.launchTargetPath}",
      "MIMode": "gdb",
      "miDebuggerPath": "${command:vscode-embedded.st.gdb}",
      "miDebuggerServerAddress": "localhost:3333",
      "debugServerPath": "${command:vscode-embedded.st.gdbserver}",
      "debugServerArgs": "--stm32cubeprogrammer-path ${command:vscode-embedded.st.cubeprogrammer} --swd --port-number 3333",
      "serverStarted": "Waiting for connection on port .*\\.\\.\\.",
      "stopAtConnect": true,
      "postRemoteConnectCommands": [
        {
          "text": "load build/debug/build/ModbusSlave.elf"
        }
      ],
      "logging": {
        "engineLogging": true
      },
      "preLaunchTask": "Build",
      "svdPath": "${command:vscode-embedded.st.svd}/STM32F401.svd"
    }
  ]
}
`

- Logs from the Embedded Tools Output Window 
`2025-01-01T22:46:54.624Z [info] [vscode-embedded] Activating extension
2025-01-01T22:47:59.920Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker
2025-01-01T22:47:59.920Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker
2025-01-01T22:47:59.921Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker
2025-01-01T22:47:59.921Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker
2025-01-01T22:47:59.980Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker
2025-01-01T22:47:59.981Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker
2025-01-01T22:47:59.981Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker
2025-01-01T22:47:59.981Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker
2025-01-01T22:48:00.000Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker
2025-01-01T22:48:00.001Z [error] [vscode-embedded] activeDebugSession was undefined in createDebugAdapterTracker`

- Logs from the Debug Console 

**Screenshots**

[image]

**Additional context**

susano254, 2025-01-01:
also apparently a few other people having the same issue 
https://stackoverflow.com/questions/79314771/unable-to-start-st-link-gdb-server-debug-stm32-board-in-vscode

uprtdev, 2025-02-03:
Seems like the debugger output has been changed in a new ST toolchain's version.

You can fix the issue locally by changing the regex in "serverStarted" parameter in `launch.json` file inside `.vscode` directory of your project.
For example, a regex that should work with both old and new versions of ST toolchains is 
`"serverStarted": "(Waiting for debugger connection.*)|(Waiting for connection on port .*\\.\\.\\.)",`.

It will be set as default template in the new version of extension once it is released.

closed 2025-02-03 by susano254